unixabg/cryptmypi #12: download improvement

superuser5, 2020-06-03:
want to suggest to modify hooks/1000-encryptpi.hook
to use torrent to download
	wget ${_IMAGEURL} -O ${_BASEDIR}/config/${_IMAGENAME}

if our image is 
export _IMAGEURL=https://images.offensive-security.com/arm-images/kali-linux-2020.2a-rpi3-nexmon-64.img.xz.torrent

step1: download it using aria2c:
   **aria2c ${_IMAGEURL} -d www/**

this will automatically download torrent file and download torrent kali image to www/kali-linux-2020.2a-rpi3-nexmon-64/

step 2: torrent files have extra folder so we moving files to ${_BASEDIR}/config/:
**mv ${_BASEDIR}/config/kali-linux-2020.2a-rpi3-nexmon-64/*.* ..**

step 3: verify hash:
**shasum -a 256 -c ${_IMAGENAME}.txt.sha256sum**

step 4: check results and exit if failed
if [ $? != 0 ]; then
  echo '${_IMAGENAME} checksum is not valid'
  exit 1
fi

splitstrikestream, 2020-06-08:
I like the idea, specially about verifying the hash (in either case). 

Some considerations on your suggestion:
1. `aria2c` does not come installed by default on kali (don't know on other distros), so I think its a good idea to keep the original download method (at least as a backup). The image url could be checked if has a torrent extension or not, so the correct download method could be used.
2. Moving the image (step 2) should be done only after verifying hash.
3. When verifying the hash, if failed it should cleanup as well. no reason to keep bad image
4. Moving should be done to the correct directory at `files/images` and not to the config. The new default location for images allow sharing between multiple configs.

As you've said, the downloaded torrent contains extra files. This file structure is standarized between distros or each torrent may have the image placed differently?

unixabg, 2020-11-24:
Greetings,
I think there may be to many specific issues since we are attempting to support various OS like Kali, PiOS, and Debian. Hence with the current download models we hare achieving a common denominator for downloading and verifying. Hence I am going to just mark closed since we have implemented the download hash verification.

closed 2020-11-24 by unixabg