unixabg/cryptmypi #13: Failed to mount /dev/mapper/crypt as root file system

superuser5, 2020-06-05:
Hello,
thank you for the great project.

I am testing RPi4 and I get the following error 
config: export _KERNEL_VERSION_FILTER="l+"
everything else is default.

both stages were successfully completed.

When i connect RPi to network and connect power, I ssh to dropbear and enter my luks passphrase and press enter:

**unlocking through ssh:**
ssh root@192.168.0.100 -p 2222 -i ~/.ssh/id_rsa                                                                                                            
Enter passphrase for /dev/mmcblk0p2: 
Connection to 192.168.0.100 closed.

**looking at the monitor, have the following output:**
Begin: Starting dropbear ... [ 11.4410601 random: dropbear: uninitialized urandom 
20 .6747371 random: dropbear: uninitialized urandom read (32 byte
30 .0756261 random: dropbear: uninitialized urandom read (32 sg eben
30 .1682271 random: dropbear: uninitialized urandom read (32 bytes read)
30 .1951641 random: cryuptsetup: uninitialized urandom read (4 bytes read)
58 .0970761 random: cryptsetup: uninitialized urandom read (2 bytes read)
58 .1084071 device—mapper: crypt: cbc(aes) using implementation “cbc—aes—neonbs”
Terminated
done.
Begin: Running /scripts/local—premount ... done.
Warning: fsck not present, so skipping root file system
mount: mounting “/dev/mapper/crypt on /root failed: Invalid argument
Failed to mount /dev/mapper/crypt as root file system.
error: unexpectedly disconnected from boot status daemon

ERROR: no password supplied
Nothing to read on input.
BusyBox v1.30.1 (Debian 1:1.30.1-4) built-in shell (ash)
Enter “’help’ for a list of built-in commands

(initramfs)

Is anything i am doing wrong?

unixabg, 2020-06-05:
Greetings,
Could you try the next-4.x branch? It is the next iteration and very close to moving to master and getting tagged for release.

superuser5, 2020-06-05:
Hello, I have used next-4.x branch, sorry i forgot to mention this.

superuser5, 2020-06-05:
I used the latest following image, testing right now on RPi 4 the examples without any changes. 
export _IMAGEURL=https://images.offensive-security.com/arm-images/kali-linux-2020.2a-rpi3-nexmon-64.img.xz

unixabg, 2020-06-06:
Greetings,
I have not tested that image yet. On another issue I reported that I have a working example for kali encrypted basic dropbear. I will label this enhancement to test newer image.

superuser5, 2020-06-06:
i got latest version of the cryptmypi repo and on the machine running native kali i run the tool to create sdcard with kali basic encrypted profile without changes, so it uses v8+ kernel for RPi4 model B.
on start, there is no network and my on "Please unlock disk crypt:" have the following error:
1. Please unlock disk crypt: *********
2. Cannot initialize device mapper, is dm_mod module loaded?
 Cannot use device crypt, name is invalid or still in use. 
cryptsetup: ERROR: crypt: cryptsetup failed, bad password or options?"

I also tested the the kali-unencrypted and 1st time kali loaded and after each following boot, tried 5 times, i get kernel panic during load, cannot load ext4 partition, i have screenshot i can provide.

Maybe i am doing something wrong, but i have eliminated all the variables and moved from VM and profile as is with the 2020.1a image.
thanks

splitstrikestream, 2020-06-08:
Hey!
I don't have a RPi 4 but will try to help you.

------------------------------------------

 > export _KERNEL_VERSION_FILTER="l+"

Re4son provides several kernels. Reading the docs here, for a 64bit image in RPi4 you should stick to "8l+"

When you tried `kali-unencrypted` you've encountered kernel panic. Could we start by trying it using the 8l+ kernel filter?

-------------------------------------------

When using an encrypted config

> mount: mounting “/dev/mapper/crypt on /root failed: Invalid argument
> Failed to mount /dev/mapper/crypt as root file system.

and 

> Cannot initialize device mapper, is dm_mod module loaded?
> Cannot use device crypt, name is invalid or still in use.
> cryptsetup: ERROR: crypt: cryptsetup failed, bad password or options?"

indicate an error while preparing the image. Can you provided the entire output for the cryptmypi script?

superuser5, 2020-06-10:
Hi,
Before i was using vmware clean kali and booted clean kali from USB live and presistance (latest install), but still didn't work.

I installed clean kali on the harddrive and was able to create and load both kali-encrypted-basic  and kali-encrypted-basic-dropbear.

the kali-encrypted-basic-dropbear failed 1st time to boot, so i created bootable sd card 3 times and last one was successfull - working remote unlock, password authentication using console and starting kali OS.

I am attaching logs for the successfull creation and non successfull. i see some errors, but not sure.

[something broke - could not boot correctly  to bearssh - kali-encrypted-basic-dropbear - RPi4 v8l+.txt](https://github.com/unixabg/cryptmypi/files/4757390/something.broke.-.could.not.boot.correctly.to.bearssh.-.kali-encrypted-basic-dropbear.-.RPi4.v8l%2B.txt)
[Successfull -  kali-encrypted-basic-dropbear - RPi4 v8l+.txt](https://github.com/unixabg/cryptmypi/files/4757391/Successfull.-.kali-encrypted-basic-dropbear.-.RPi4.v8l%2B.txt)

I could not get logs from the previous failures, as rrun installation many times, but i can rerun if it will be useful. i hope this helps.