unixabg/cryptmypi #14: unlocking error on RP3B+ - Not enough available memory to open a keyslot

superuser5, 2020-06-05:
hello,

thank you for much for tool. I am testing the next-4.x branch if that's ok.

I am testing it on the RPi3 Raspberry Pi 3 B+ (B Plus)  with 1.4 GHz 64-bit Quad-Core Processor, 1 GB RAM + 32gb sd card.
 
i use config export _KERNEL_VERSION_FILTER="v8+" and everything else default, config is below.
 
when i unlock remotely i get the following error
kali@kali ~> ssh root@192.168.0.21 -i /home/kali/.ssh/burner24 -p 2222
The authenticity of host '[192.168.0.21]:2222 ([192.168.0.21]:2222)' can't be established.
ECDSA key fingerprint is SHA256:Q6viG9yNfjVXFfrTcDIxQ3zJmricBXUQxiGwTYNHfzU.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '[192.168.0.21]:2222' (ECDSA) to the list of known hosts.
Enter passphrase for /dev/mmcblk0p2: 
**Not enough available memory to open a keyslot.**
Connection to 192.168.0.21 closed.

i tried googling for the answer and cannot find anything, is something i can fix in the tool?

**here the config used:**
export _KERNEL_VERSION_FILTER="v8+"
export _HOSTNAME="pikali"
export _BLKDEV="/dev/sdb"
export _LUKSCIPHER="aes-cbc-essiv:sha256"
export _LUKSPASSWD="luks"
export _IMAGEURL=https://images.offensive-security.com/arm-images/kali-linux-2020.2a-rpi3-nexmon-64.img.xz
                  
export _PKGSPURGE=""
export _PKGSINSTALL="macchanger"
export _SSH_LOCAL_KEYFILE="/home/kali/.ssh/key"
stage1_hooks(){
    myhooks "stage1"
}
stage1_optional_hooks(){
    myhooks "optional-initramfs-luksnuke"
    myhooks "optional-sys-gpugovernor-ondemand"
    myhooks "optional-sys-dns"
}
_SETUP_SCRIPT='setup.sh'
stage2_optional_hooks(){
    myhooks "optional-sys-rootpassword"
}
export _LUKSNUKEPASSWD="nuke"
export _DNS1='1.1.1.1'
export _DNS2='8.8.8.8'
export _ROOTPASSWD="toor"

thank you in advance

unixabg, 2020-06-06:
Greetings,
I just pushed to next-4.x and master a kali encrypted basic dropbear example that builds and unlocks on my test setup. Please try that example as a starting point and see if you have better luck. If so please close issue.

splitstrikestream, 2020-06-08:
I've encountered this error once, but it never happened again (even on the same image). 

To prevent eventually getting stuck some other time, Ive placed a command to reboot the Pi if a problem like this happened, so if it happened, in a few seconds everything should be back and you could retry. 

Do you still have this image? Can you verify it is a deterministic error?

superuser5, 2020-06-08:
Hi,
i have tested on RPi3B+ and no luck to unlock it :(

On the client side i have this, i have to enter password twice for some reason, even though i have same error while using bearssh "Not enough available memory to open a keyslot.", but on the screen it drops to Busybox with some errors, please find screenshot attached.

**ssh console**
kali@kali ~> ssh root@192.168.1.13 -p 2222 -i /home/kali/.ssh/id_rsa_test 
root@192.168.1.13: Permission denied (publickey).
kali@kali ~ [255]> ssh root@192.168.1.13 -p 2222 
Enter passphrase for /dev/mmcblk0p2: 
Not enough available memory to open a keyslot.
Connection to 192.168.1.13 closed.
kali@kali ~> ssh root@192.168.1.13 -p 2222
Enter passphrase for /dev/mmcblk0p2: 
Not enough available memory to open a keyslot.
Connection to 192.168.1.13 closed.

and the screenshot is attached.
[image]

superuser5, 2020-06-08:
I have tested the RPI4 and similar problem, even thought i don't get any errors during step 1 or step2 of the process, i need to  enter password few times to get accepted and get other errors:

**ssh console**

**screenshot error testing kali unencrypted basic dropbear**
 
[image]

superuser5, 2020-06-08:
> I've encountered this error once, but it never happened again (even on the same image).
>
> To prevent eventually getting stuck some other time, Ive placed a command to reboot the Pi if a problem like [...]

Hello,
I tried imaging multiple times  but i still have errors above on the kali-encrypted-basic-dropbear and pios-encrypted-basic and test on RPi3B+ and RPI4. I tried imaging multiple times, make sure there is enough space and for some reason it does not recognize my password consistently on both and have "Not enough available memory to open a keyslot" on RPi3 consistently.

Is there a way to verify that there is no corruption during image creation process?

unixabg, 2020-06-08:
Greetings,

> Hello,
> I tried imaging multiple times but i still have errors above on the kali-encrypted-basic-dropbear and pios-encrypted-basic and test on RPi3B+ and RPI4. I tried imaging multiple times, make [...]

The only thing I can think of for you since the examples/kali-encrypted-basic-dropbear works for me is to try building inside a pristine VM with latest kali (I am using kali-linux-2020.2-installer-amd64.iso installer with VirtualBox) and a fresh clone. I will leave this open and wait for you to test in a pristine environment.

superuser5, 2020-06-12:
on the same machine that successfully creates RPI4b images, i run the following profile as is for RPi3B+: $ sudo ./cryptmypi.sh examples/kali-encrypted-basic-dropbear | tee rpi3b_plus_kali-encrypted-basic-dropbear_build_log.txt

remote and local unlock gives same error message:
$ ssh root@192.168.1.13 -p 2222
Enter passphrase for /dev/mmcblk0p2: 
Not enough available memory to open a keyslot.
Connection to 192.168.1.13 closed.

log file attached.
[rpi3b_plus_kali-encrypted-basic-dropbear_build_log.txt](https://github.com/unixabg/cryptmypi/files/4767968/rpi3b_plus_kali-encrypted-basic-dropbear_build_log.txt)

superuser5, 2020-06-12:
tested same RPi3 with the example profile as is (running v7+) 
sudo ./cryptmypi.sh examples/pios-encrypted-basic | tee pios-encrypted-basic_build-log.txt.

**Problem**: I cannot authenticate using local unlock (cryptsetup: ERROR: crypt: cryptsetup failed, bad password or options?)

log file:
[pios-encrypted-basic_build-log.txt](https://github.com/unixabg/cryptmypi/files/4768062/pios-encrypted-basic_build-log.txt)

splitstrikestream, 2020-06-14:
The original issue **Not enough available memory to open a keyslot
** should be solved in the latest PR #20

unixabg, 2020-06-18:
Greetings,
I am going to close this since I tested latest v4-1-beta tagged release with both rpi3b+ and rpi4 on kali-encrypted-basic-dropbear and both runs worked as expected. If more assistance is needed please open a new issue.

closed 2020-06-18 by unixabg

howtodo1, 2020-10-05:
I have the same problem and same kind of raspberry pi, did I do something wrong during the build?

unixabg, 2020-10-25:
Greetings,
First thank you for the update and information. I will be on the look out for this during my testing in the coming weeks. Please know that I am really only testing against the next-4.x branch and primary testing on rpi3b+.