unixabg/cryptmypi #18: Unlocked succesfully but "fsck not present, so skipping root file system"

Ud0g-Py, 2020-06-10:
Hi! Congratulations for the great project.

Im trying to get it working but im having some troubles.

I have a Rpi4. 
Using `kali-encrypted-basic-nuke`, conf file is by default except by `export _KERNEL_VERSION_FILTER="8l+"`, `export _LUKSPASSWD="xxxx"`, `export _ROOTPASSWD="xxxx"`, `export _LUKSNUKEPASSWD="xxxx"`.

Im using my Rpi on local so i dont need ssh. At boot, it prompt the crypt pass. This is the output:

```
cryptsetup: crypt: set up successfully
done.
Begin: Running /scripts/local-premount ... done.
Warning: fsck not present, so skipping root file system
mount: mounting /dev/mapper/crypt on /root failed: Invalid argument
Failed to mount /dev/mapper/crypt as root file system.

BusyBox v1.30.1 (Debian 1:1.30.1-4) built-in shell (ash)
Enter 'help' for a list of built-in commands.
(initramfs)
```

Ud0g-Py, 2020-06-11:
Hi again.

> Hi,
> 
> I had simmilar issues and something was failing during the build process. yesterday i was able to build kali-encrypted-basic-dropbear and kali-encrypted-basic on the laptop running kali. theck the output for any errors.
> 
> Could you paste the full output?
> 
> thanks

Tested right now with `kali-encrypted-basic` and with the default img from .conf file and worked like a charm.

Im going to try replicate the error. In a few hours ill post the logs.

splitstrikestream, 2020-06-14:
Latest PR #20 fixes issues I found when using `kali-linux-2020.2a-rpi3-nexmon-64.img.xz`. You should try it when its accepted. As a reminder, keep the `export _KERNEL_VERSION_FILTER="8l+"`

Ud0g-Py, 2020-06-15:
Sorry for delay.

> Latest PR #20 fixes issues I found when using `kali-linux-2020.2a-rpi3-nexmon-64.img.xz`. You should try it when its accepted. As a reminder, keep the `export _KERNEL_VERSION_FILTER="8l+"`

Splitstrikestream is right. `kali-linux-2020.2a-rpi3-nexmon-64.img.xz`is the problem. 

[kali-encrypted-basic-nuke_2a.txt](https://github.com/unixabg/cryptmypi/files/4783466/kali-encrypted-basic-nuke_2a.txt)

This is the log. 

I tried with 1a & 1b .img and everything is going well.

unixabg, 2020-06-18:
Greetings,
Latest v4.1-beta uses 2b and test runs show all well. So marking completed for now.

closed 2020-06-18 by unixabg