unixabg/cryptmypi #21: cant boot after first 'apt update'

Ud0g-Py, 2020-06-16:
Hi again.

I dont know if I should open another issue, but I have problems after make first `apt update && apt upgrade`.

Everything go well until reboot. 

LUKS password is not working anymore. Key cant be recognized, same output as if I enter a wrong password.

I tried to copy the content of `boot` partition from a fresh install and I can log again, but if i make `startx` on tty1, system freeze and cant be recovered. 

Image used is `2020.1a`

howtodo1, 2020-10-05:
I have a similar problem, when I boot it up I am surprised to see that it works but after I put in the LUKS password it says there isn't anymore available memory to open a key slot and I don't know how to fix it.

unixabg, 2020-10-30:
Greetings,
Not sure on this. I just tested on latest 4.3-beta release and kali-encrypted-basic-dropbear and could not reproduce the issue. Going to mark closed for now.

closed 2020-10-30 by unixabg