unixabg/cryptmypi #22: Initramfs-wifi issues

Botts85, 2020-07-09:
I have tried to use the "Complete" example a couple of times to create a headless Kali RPI0w build.

Each time I run into issues with initramfs wifi.

The common error messages are: 

```
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
sh: wpa_state=COMPLETED: unknown operand
Success: WLAN Online
```

Then:

```
ipconfig: wlan0: SIOCGIFINDEX: No such devices
ipconfig: no devices to configure
```

Then:

```
/scripts/init-premount/a_enable_wireless: .: line 275: can't open 'run/net-wlan0.conf': No such file or directory
done.
```

Any ideas on where to start for troubleshooting?  I started from scratch with the config file, and still have the same result.

unixabg, 2020-07-11:
Greetings,

Would you be able to share your conf file? If so please scrub any sensitive information prior to sharing.

unixabg, 2020-07-12:
Greetings,

I have tested the wifi by adding wifi settings and hook to examples/kali-encrypted-basic. The wifi does work on my rpi3 b+. I do not have a rpi0w to test with. Perhaps you could try something like what I have done with the rpi0w and see if you have any success.

Botts85, 2020-07-12:
This is the config file I was using:

```
###############################################################################
## cryptmypi profile ##########################################################

# EXAMPLE OF A ENCRYPTED KALI CONFIGURATION, USING ALL AVAILABLE OPTIONAL HOOKS
#   Will create a encrypted Kali system:
#   - "optional-sys-gpugovernor-ondemand" that changes the cpugovernor to
#       ondemand, specially usefull when the RPi will run using batteries
#
#   - during boot the encryption password will be prompted
#   - the luks password unlock can be done thorugh ssh (dropbear)
#   - the dorpbear ssh port will be exposed to the internet using sshhub.de
#       (reverse ssh jump thorugh sshhub.de server)
#   - wifi will be configured in initramfs so that the RPi does not need wired
#       connection (even at the boot unlocking ecnryption process at initramfs)
#   - a luks nuke password is configured, just in case
#
#   - ssh server will available after boot
#       The id_rsa.pub public key will be added to authorized_keys
#   - "optional-sys-dns" that preconfigures DNS to google (see vars bellow)
#   - "optional-sys-sshhub" that exposes the ssh server to the internet
#       thorugh sshhub.de using reverse ssh jumps, eliminating the neet
#       of router port forarding (after boot). Local ssh config is changed so
#       alias is defined and simplifies connection (and usage of other tools,
#       such as git, rsync, etc).
#   - "optional-sys-rootpassword" that sets root password
#   - "optional-sys-wifi" that pre configures wifi on system (after boot)
#   - "optional-sys-vpnclient" configures vpn
#
#   - IODINE hook is not enabled, but may work. Not tested, work may be needed
#       on it, as initrafs and sys configs are not separated. It would tunnel
#       traffic thorugh DNS, bypassing most firewalls.

# General settings ------------------------------------------------------------
# You need to choose a kernel compatible with your RPi version.
#   Kali RPi images name its kernels:
#   - Re4son+ is for armv6 devices (ie. RPi1, RPi0, and RPi0w)
#   - v7+ and v8+ sufixes are for the 32bit and 64bit armv7 devices (ie. RPi 3)
#   - l+ sufix in the name means they will be ready for the RPi4.
export _KERNEL_VERSION_FILTER="Re4son+"

# HOSTNAME
#   Each element of the hostname must be from 1 to 63 characters long and
#   the entire hostname, including the dots, can be at most 253
#   characters long.  Valid characters for hostnames are ASCII(7) letters
#   from a to z, the digits from 0 to 9, and the hyphen (-)
export _HOSTNAME="rpi"

# BLOCK DEVICE
#   The SD card or USD SD card reader block device
#   - USB drives will show up as the normal /dev/sdb, /dev/sdc, etc.
#   - MMC/SDcards may show up the same way if the card reader is USB-connected.
#   - Internal card readers normally show up as /dev/mmcblk0, /dev/mmcblk1, ...
#   You can use the lsblk command to get an easy quick view of all block
#   devices on your system at a given moment.
export _BLKDEV="/dev/mmcblk0"

# LUKS ENCRYPTION -------------------------------------------------------------
## Encryption Cypher
export _LUKSCIPHER="aes-cbc-essiv:sha256"

## Encryption Password
export _LUKSPASSWD="passwordtest"

# LINUX IMAGE FILE ------------------------------------------------------------
export _IMAGEURL=https://images.offensive-security.com/arm-images/kali-linux-2020.2a-rpi0w-nexmon.img.xz
export _IMAGESHA="c760d29e054744352b489e5b6168954addbb0b597aac67cdeab24f9c06fba7c0"

[... 160 lines cut ...]
#     For reference, each Raspberry PI version's default wifi chipset:
#     - RPi0W     bcm43438 ou bcm43430a¹
#     - RPi3b     bcm43430a¹
#     - RPi3b+    bcm43455c0
# Optional variable (default:'brcmfmac brcmutil cfg80211 rfkill')
# export _INITRAMFS_WIFI_DRIVERS='brcmfmac brcmutil cfg80211 rfkill'

# ROOT PASSWORD CHANGER settings ----------------------------------------------
# Hooks
#   optional-sys-rootpassword
#       Changes the system root password

## The new root password
export _ROOTPASSWD="kaliroot"

# OPENVPN settings ------------------------------------------------------------
# Hooks
#   optional-sys-openvpn
#       Attemps to configure the system to use an VPN (as a client)

## Zip file containing 'client.conf' and any other deps (creds, certs, etc)
# Optional variable (default: "openvpn.zip")
export _OPENVPN_CONFIG_ZIP="openvpn.zip"

```

Botts85, 2020-07-13:
So I tried it again with both complete and encrypted-basic, and it's having issues building initramfs with wifi.
The output is here:

```
- Calling 7000-stage2-initramfs.hook ...
Attempting to build new initramfs ... (CHROOT is /mnt/cryptmypi)
  Creating symbolic links from current physical device to crypttab device (if not using sd card mmcblk0p)
  Using kernel '4.19.93-Re4son+'
  Building new initramfs ...
Adding module /lib/modules/4.19.93-Re4son+/kernel/drivers/md/dm-mod.ko
Adding module /lib/modules/4.19.93-Re4son+/kernel/drivers/md/dm-crypt.ko
Adding module /lib/modules/4.19.93-Re4son+/kernel/net/rfkill/rfkill.ko
Adding module /lib/modules/4.19.93-Re4son+/kernel/net/wireless/cfg80211.ko
Adding module /lib/modules/4.19.93-Re4son+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmutil/brcmutil.ko
Adding module /lib/modules/4.19.93-Re4son+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4373-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4356-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4354-sdio.bin for module brcmfmac
Adding firmware /lib/firmware/brcm/brcmfmac43455-sdio.bin
Adding firmware /lib/firmware/brcm/brcmfmac43430-sdio.bin
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43430a0-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4339-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43362-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4335-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43341-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43340-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4334-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4330-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4329-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43241b5-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43241b4-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43241b0-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43143-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4373.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43569.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43242a.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43236b.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43143.bin for module brcmfmac
Copying module directory kernel/drivers/usb/dwc2

```

unixabg, 2020-07-17:
Greetings,

I do not have like hardware to test with but does adjusting the _INITRAMFS_WIFI_DRIVERS variable as below make any difference?

```
## Modules to load at the initramfs, add your wifi firmware and drivers here
#     For reference, each Raspberry PI version's default wifi chipset:
#     - RPi0W     bcm43438 ou bcm43430a¹
#     - RPi3b     bcm43430a¹
#     - RPi3b+    bcm43455c0
# Optional variable (default:'brcmfmac brcmutil cfg80211 rfkill')

export _INITRAMFS_WIFI_DRIVERS='brcmfmac brcmutil cfg80211 rfkill bcm43438 ou bcm43430a'

```

unixabg, 2020-10-30:
Going to close for now since I can not reproduce due to lack of hardware.

closed 2020-10-30 by unixabg