unixabg/cryptmypi #23: Scripts fails due to not being able to resolve hosts.

KoalaV2, 2020-09-01:
As said in title the scripts fails when it's trying to install dropbear because it cannot resolve hosts, my guess would be incorrect DNS name server names in the /etc/resolv.conf file, I've got no clue how to fix it with the script so sadly cant make any pull requests.

KoalaV2, 2020-09-01:
Here's the output: `Removing current build files...
# Executing both stages #######################################################
###############################################################################
                               C R Y P T M Y P I
                               ---- Stage 1 ----
v4.1-beta
###############################################################################

--- Custom STAGE1 SELECTED
    stage1profile_complete

--- Executing:

Attempting to run stage1 hooks ...
- Calling 1100-stage1-sanity-qemu.hook ...
    ... 1100-stage1-sanity-qemu.hook completed!

- Calling 2000-stage1-image-download.hook ...
ARM image kali-linux-nightly-rpi3-nexmon-64-2020-09-01-06-41.img.xz already exists!
Validating image checksum...
Done: Image is valid!
    ... 2000-stage1-image-download.hook completed!

- Calling 2100-stage1-image-extract.hook ...
Extracting image: kali-linux-nightly-rpi3-nexmon-64-2020-09-01-06-41.img.xz ...
Mounting loopback ...
Syncing image/root to /home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root ...
Syncing image/root/boot to /home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root/boot ...
Cleaning loopback ...
    ... 2100-stage1-image-extract.hook completed!

- Calling 2200-stage1-setup-chroot.hook ...
Preparing RPi chroot mount structure at '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root'.
Mounting '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root/dev/' ...
Mounting '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root/dev/pts' ...
Mounting '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root/sys/' ...
Mounting '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root/proc/' ...
Updating apt-get
Err:1 http://kali.download/kali http://http.kali.org/kali InRelease    
  Temporary failure resolving 'kali.download'
Err:2 http://http.re4son-kernel.com/re4son kali-pi InRelease           
  Temporary failure resolving 'http.re4son-kernel.com'
Reading package lists... Done                                          
W: Failed to fetch http://kali.download/kali/dists/http://http.kali.org/kali/InRelease  Temporary failure resolving 'kali.download'
W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  Temporary failure resolving 'http.re4son-kernel.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Updating sytem packages
    ... 2200-stage1-setup-chroot.hook completed!

- Calling 2300-stage1-locale.hook ...
Configuring system locale...
_LOCALE not set, using default value 'en_US.UTF-8'
Uncommenting locale 'en_US.UTF-8' for inclusion in generation
Updating /etc/default/locale
Installing locales
- Installing locales
Reading package lists... Done
Building dependency tree       
Reading state information... Done
locales is already the newest version (2.30-8).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Updating env variables
+ export LC_ALL=en_US.UTF-8
+ export LANG=en_US.UTF-8
+ LANG=en_US.UTF-8
+ export LANGUAGE=en_US.UTF-8
+ LANGUAGE=en_US.UTF-8
(Re)Generating locale
Generating locales (this might take a while)...
  en_US.UTF-8... done   
[... 50 lines cut ...]
  Temporary failure resolving 'http.re4son-kernel.com'
Err:2 http://kali.download/kali http://http.kali.org/kali InRelease    
  Temporary failure resolving 'kali.download'
Reading package lists... Done   
W: Failed to fetch http://kali.download/kali/dists/http://http.kali.org/kali/InRelease  Temporary failure resolving 'kali.download'
W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  Temporary failure resolving 'http.re4son-kernel.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
+ apt-get -y install dropbear
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package dropbear
+ exit 1
ERROR: Could not install dropbear correctly... Exiting.
Tearing down RPi chroot mount structure at '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root'.
Unmounting binds
umount: /dev/mmcblk0: not mounted.
umount: /dev/mmcblk0p1: not mounted.
umount: /dev/mmcblk0p2: not mounted.
umount: /mnt/cryptmypi: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /dev/mapper/crypt: no mount point specified.
Device crypt is not active.
~/cryptmypi $ 
`

unixabg, 2020-09-02:
Greetings,

First let me say thank you for the report and testing with the project. Would you be able to attempt a build using the next-4.x branch? Also if so please do a git pull --rebase and note that if your config used a setup.sh script that will need renamed to setup-stage1.sh. I believe the issue you are encountering is resolved in the next-4.x branch. Please report back and close if that resolves your issue.

KoalaV2, 2020-09-03:
Hey, yeah absolutely, I'll try it when I get home in a couple of hours and
report back

BMXT, 2020-09-03:
Hi, I get the same problem using the kali-complete example and next-4.x branch.
Tested on Ubuntu 20.04.1.
[image]

ebourmalo, 2020-09-03:
Hi,

I fixed this issue on my fork, didn't have time to make a pull request.
You can update the `chroot_update` method in `chroot.fns` with this (fix is commented):

```bash
chroot_update(){
    [ -z "${CHROOTDIR}" ] && {
        echo_error "Chroot dir was not defined! Aborting..."
        exit 1
    }

    echo_debug "${CHROOTDIR}"
    echo_debug "Updating apt-get"
    # -----> Fix expired keys of repo http://http.re4son-kernel.com/re4son
    chroot ${CHROOTDIR} apt-key adv --keyserver keys.gnupg.net --recv-keys 11764EE8AC24832F
    chroot ${CHROOTDIR} apt-get update

    echo_debug "Updating sytem packages"
    # chroot ${CHROOTDIR} apt-get -y upgrade
}
```

The issue was reported here: https://www.reddit.com/r/Kalilinux/comments/ij7trs/unable_to_update_kali_on_pi/

ebourmalo, 2020-09-03:
I also had to run the hook `optional-sys-dns` to set the DNS before `chroot_update` otherwise, you can't download a thing.

File hooks/2200-stage1-setup-chroot.hook
```bash
chroot_mount ${_BUILDDIR}/root
# Run hook DNS to fix issues "Temporary failure resolving..."
myhooks "optional-sys-dns"
chroot_update
```

With these 2 updates, you'll be good for the fetch of the repos.
Hope this helps

unixabg, 2020-09-03:
Greetings,

I just finished running a test build with latest next-4.x with examples/kali-encrypted-basic and all appears in good order. The commit that addresses the case is 95a4680 .  Again if you are failing on next-4.x branch please verify you are up to date with git pull --rebase.

KoalaV2, 2020-09-03:
> Greetings,
>
> I just finished running a test build with latest next-4.x with examples/kali-encrypted-basic and all appears in good order. The commit that addresses the case is [...]

Hello only updating to the new commit does not help the issue. even including the optional-sys-dns like in the kali-full example

KoalaV2, 2020-09-03:
> 
> 
> I also had to run the hook `optional-sys-dns` to set the DNS before `chroot_update` otherwise, you can't download a thing.
> 
> File hooks/2200-stage1-setup-chroot.hook
> 
> ```shell
> chroot_mount ${_BUILDDIR}/root
> # Run hook DNS to fix issues "Temporary failure resolving..."
> myhooks "optional-sys-dns"
> chroot_update
> ```
> 
> With these 2 updates, you'll be good for the fetch of the repos.
> Hope this helps

although while this doesnt seem to work it atleast gives me some progreess, i now get that "not found 104.18.103.100 80" on the kali.download/kali Release repo and then i get that it does not have a release file so apt update and install dropbear *still* cancells. `apt-get failed: Trying to recover...
+ sleep 5
+ apt-get update
Ign:1 http://kali.download/kali http://http.kali.org/kali InRelease
Err:2 http://kali.download/kali http://http.kali.org/kali Release
  404  Not Found [IP: 104.18.103.100 80]
Hit:3 http://http.re4son-kernel.com/re4son kali-pi InRelease
Reading package lists... Done
E: The repository 'http://kali.download/kali http://http.kali.org/kali Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
+ apt-get -y install dropbear
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package dropbear
+ exit 1
ERROR: Could not install dropbear correctly... Exiting.
Tearing down RPi chroot mount structure at '/home/koala/cryptmypi/examples/kali-encrypted-basic-dropbear//build/root'.
Unmounting binds
umount: /dev/mmcblk0: not mounted.
umount: /dev/mmcblk0p1: not mounted.
umount: /dev/mmcblk0p2: not mounted.
umount: /mnt/cryptmypi: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /dev/mapper/crypt: no mount point specified.
Device crypt is not active.
~/cryptmypi $`

unixabg, 2020-09-03:
Greetings,

Please note that the commit I listed checks for resolv.conf and if one does not exist, then it populates one. So should be no need unless you want to override see hooks/0300-preconditions-dns.hook which populates the DNS variables that I use in that commit if not already set. I think the issue is that you are hitting now is a repo that has an issue.  For now I am going to close this issue.

closed 2020-09-03 by unixabg