unixabg/cryptmypi #26: cryptsetup failed,bad password or options

git-chenjiqing, 2020-10-30:
cryptsetup failed,bad password or options
for pi4
cryptmypi-4.1-beta
./cryptmypi.sh examples/pios-encrypted-basic

unixabg, 2020-10-30:
Greetings,

I just released 4.3-beta. Please try another run with the latest version.

git-chenjiqing, 2020-11-10:
Hi, how to implement SSH Remote decryption after pi4 encryption?

unixabg, 2020-11-23:
Greetings,

So the I had a chance to do some testing and the examples/pios-encrypted-basic with _KERNEL_VERSION_FILTER="l+" worked as expected. I also made a commit example for examples/pios-encrypted-basic-dropbear on the next-4.x branch.

_KERNEL_VERSION_FILTER="v7+" tested on rpi3b+ and working
_KERNEL_VERSION_FILTER="v8+" tested on rpi3b+ and **NOT working**
_KERNEL_VERSION_FILTER="l+"  tested on rpi4b with 4gb ram and working

General remote unlock would be something like:
1. Know the pi ip address (I have a monitor hooked up on my test setup)
2. ssh to unlock the pi with ssh password on key first, and unlock on pi next:

ssh -p2222 root@rpi.ip.address

You should see something like:

Enter passphrase for key '/home/user/.ssh/id_rsa': 

And then unlock rpi luks:

Enter passphrase for /dev/mmcblk0p2: 

I hope this information helps you. I will leave open for a bit but in general I think this issue has been addressed.

unixabg, 2020-12-05:
Greetings,
No response and I think the example as said addressed this issue.

closed 2020-12-05 by unixabg