unixabg/cryptmypi #27: ARM Linux Full Disk Encryption/Decryption Without Password/Dropbear/SSH

alimp5, 2020-11-11:
Drear Richard Nelson,
Greetings!

Richard, I need to secure my Python and Bash codes and other things (Folders, Partisions, ... <==> Full Disk Encryption) on MicroSD and/or eMMC storage cards.

I don't want to follow the Dropbear/SSH method for the decryption process. I need a non-interactive (without user interaction) encryption/decryption method by the device itself.
I want the device (RPi-3) itself to check and validate hardware information (CPU/HDD Serials, MACs, etc...) and then auto to decrypt/encrypt and boot-up the device without user interaction.

i'm thankful if you help me and send me useful docs, guides, videos, practical tutorials,.............. :(

I have these devices:  NanoPi M4,  a RPi-3 B,    a RPi-3 B+,   a MicroSD / eMMC  Card.

I read the instructions and tutorials below. But my need is something else :(
https://www.kali.org/docs/arm/raspberry-pi-with-luks-disk-encryption/
https://www.kali.org/docs/arm/raspberry-pi-full-encryption/
https://www.kali.org/tutorials/nuke-kali-linux-luks/
https://kali.training/topic/exercise-chapter-4-installing-kali-linux/
https://github.com/unixabg/cryptmypi

Thanks a lot.........

unixabg, 2020-11-13:
Greetings,

First, I _think_ what you want is doable. Second, I have a couple of requests for a couple of custom things. Hence I am considering a folder named other for such use cases with small examples/snippets/pseudocode that users may find helpful. I will  label as enhancement and work on getting that folder in place with some pseudocode I have already. That code may even move your idea forward.

unixabg, 2020-11-13:
Greetings,

I made a sample commit in next-4.x branch of fbe234bd0ae which has pseudocode of unlock from usb. I will leave this open for the time being as an enhancement and contrib of psuedocode or actual production code in other folder welcome.

alimp5, 2020-11-14:
I think tools below are useful:
1. `hwinfo`
2. Using /sys file system: `grep "" /sys/class/dmi/id/[pbs]*`
3. `lshw`  example: `lshw -C system`
4. `inxi -M`
5. `dmidecode -t system` `dmidecode -t 1` etc......
6. `hdparm -i /dev/sda`
7. `ls /dev/disk/by-uuid/`
8. /proc files: `cat /proc/version`

unixabg, 2020-12-16:
Greetings,
So I have thought some more about this and all you can do is make it difficult for someone to still see how we do it. I think if I wanted a small sanity check on initramfs I would do that in the stage2-otherscript.sh and would probably do something like this:

https://bbs.archlinux.org/viewtopic.php?id=189588

You should be able to modify the above to test how you want. I suppose at some point I might include this in the project, but at for now I do no see the point, given a certain skill level will be able to find out how one is unlocking if based on a hardware test in initramfs. So I hope that link helps you and I am going to close the issue.