unixabg/cryptmypi #28: sshhub.de Not work | Adding configuration for VPS

loliDe, 2020-11-12:
Hi @unixabg !
Thx for your cool stuff, but I have some problem: I want to use "examples/kali-complete" example with full encryption and remote Drop-Bear SSH via sshhub.de, **but** for Now sshhub.de doesn't work, and I cannot login and unlock my "drop-box", I begin to modify your script, but I am stuck: I cannot configured proprely work your script with My Jump SSH server, I want to use my VPS  (Ubuntu 18.04 with ssh server) as Jump SSH Server such as You used sshub.de.
Can You help me Plz....  I want to use Rpi4 with full disk encryption + Drop Bear SSH for remote unlocking + own VPS  with Jump SSH Server.
Cheers!

loliDe, 2020-11-16:
@unixabg 
Are you here??
Can you hlp..plz???

unixabg, 2020-11-17:
Greetings,
I believe you are correct that sshhub.de is not working right now. Hence, I believe your request is a good idea to migrate from a single hard coded ssh jump server (sshhub.de atm) to generic ssh jump server. I will mark as enhancement and leave open to work on as I get time.

unixabg, 2020-12-14:
Greetings,
I have made initial commit of 5caee8e2dd9 for adding configuration of remote ssh jump server. I have only tested the  hooks/0000-experimental-initramfs-sshjump.hook for remote unlock. A couple of things I would say is not sure if everyone sets up jump servers the same. On my test jump server I configured for keys only, so I added a jumpuser and public keys of devices who need access to authorized_keys. Also I had to manually register the hostname in /etc/hosts file. After that I was able to remote unlock via jump server as expected. Going to mark issue closed and if there are bugs please submit new issue with details and as always patches welcome.

closed 2020-12-14 by unixabg

loliDe, 2020-12-15:
@unixabg Thx a lot of your Great WORK!!!!