unixabg/cryptmypi #30: initramfs-wifi.hook for Raspberry Pi 4

loliDe, 2020-11-20:
Hello @unixabg
If it possible modify  initramfs-wifi.hook for Raspberry Pi 4?
With default options: "_brcmfmac brcmutil cfg80211 rfkill_" my Rpi 4 doesn't work...
Thx =)

loliDe, 2020-11-26:
@unixabg Can you help me? How I can run wifi initramfs with Rpi4B?
Plz...

unixabg, 2020-11-28:
Greetings,
So what I can say is that 0000-experimental-initramfs-wifi.hook has some issues. Please know that I will examine, but I am not sponsored for this project and only get to work on in my spare time. However, I do think that it should be possible to activate the wlan0 networking for remote unlock. The 0000-optional-sys-wifi.hook is in good order and operates as expected.

loliDe, 2020-11-29:
> Greetings,
> So what I can say is that 0000-experimental-initramfs-wifi.hook has some issues. Please know that I will examine, but I am not sponsored for this project and only get to work on in my [...]

Hi @unixabg  thx for your answer
But I can do myself, Can You help me only with wifi module:
Which wifi drivers I must to use, then start My Rpi4B, default doesn't work
_INITRAMFS_WIFI_DRIVERS='brcmfmac brcmutil cfg80211 rfkill'
Rpi4B has BCM2711, but I don't know default wifi chipset
Can you help me?

loliDe, 2020-12-01:
@unixabg 
Help, my config file
```
###############################################################################
## cryptmypi profile ##########################################################

# EXAMPLE OF A ENCRYPTED KALI CONFIGURATION, USING ALL AVAILABLE OPTIONAL HOOKS
#   Will create a encrypted Kali system:
#   - "optional-sys-gpugovernor-ondemand" that changes the cpugovernor to
#       ondemand, specially usefull when the RPi will run using batteries
#
#   - during boot the encryption password will be prompted
#   - the luks password unlock can be done thorugh ssh (dropbear)
#   - the dorpbear ssh port will be exposed to the internet using sshhub.de
#       (reverse ssh jump thorugh sshhub.de server)
#   - wifi will be configured in initramfs so that the RPi does not need wired
#       connection (even at the boot unlocking ecnryption process at initramfs)
#   - a luks nuke password is configured, just in case
#
#   - ssh server will available after boot
#       The id_rsa.pub public key will be added to authorized_keys
#   - "optional-sys-dns" that preconfigures DNS to google (see vars bellow)
#   - "optional-sys-sshhub" that exposes the ssh server to the internet
#       thorugh sshhub.de using reverse ssh jumps, eliminating the neet
#       of router port forarding (after boot). Local ssh config is changed so
#       alias is defined and simplifies connection (and usage of other tools,
#       such as git, rsync, etc).
#   - "optional-sys-rootpassword" that sets root password
#   - "optional-sys-wifi" that pre configures wifi on system (after boot)
#   - "optional-sys-vpnclient" configures vpn
#
#   - IODINE hook is not enabled, but may work. Not tested, work may be needed
#       on it, as initrafs and sys configs are not separated. It would tunnel
#       traffic thorugh DNS, bypassing most firewalls.

# General settings ------------------------------------------------------------
# You need to choose a kernel compatible with your RPi version.
#   Kali RPi images name its kernels:
#   - Re4son+ is for armv6 devices (ie. RPi1, RPi0, and RPi0w)
#   - v7+ and v8+ sufixes are for the 32bit and 64bit armv7 devices (ie. RPi 3)
#   - l+ sufix in the name means they will be ready for the RPi4.
export _KERNEL_VERSION_FILTER="v8l+"

# HOSTNAME
#   Each element of the hostname must be from 1 to 63 characters long and
#   the entire hostname, including the dots, can be at most 253
#   characters long.  Valid characters for hostnames are ASCII(7) letters
#   from a to z, the digits from 0 to 9, and the hyphen (-)
export _HOSTNAME="user1"

# BLOCK DEVICE
#   The SD card or USD SD card reader block device
#   - USB drives will show up as the normal /dev/sdb, /dev/sdc, etc.
#   - MMC/SDcards may show up the same way if the card reader is USB-connected.
#   - Internal card readers normally show up as /dev/mmcblk0, /dev/mmcblk1, ...
#   You can use the lsblk command to get an easy quick view of all block
#   devices on your system at a given moment.
export _BLKDEV="/dev/sdc"

# LUKS ENCRYPTION -------------------------------------------------------------
## Encryption Cypher
export _LUKSCIPHER="aes-cbc-essiv:sha256"

## Encryption Password
export _LUKSPASSWD="password"

# LINUX IMAGE FILE ------------------------------------------------------------
export _IMAGEURL=https://images.kali.org/arm-images/kali-linux-2020.4-rpi4-nexmon-64.img.xz
export _IMAGESHA="c6ceee472eb4dabf4ea895ef53c7bd28751feb44d46ce2fa3f51eb5469164c2c"

[... 159 lines cut ...]
## Modules to load at the initramfs, add your wifi firmware and drivers here
#     For reference, each Raspberry PI version's default wifi chipset:
#     - RPi0W     bcm43438 ou bcm43430a¹
#     - RPi3b     bcm43430a¹
#     - RPi3b+    bcm43455c0
# Optional variable (default:'brcmfmac brcmutil cfg80211 rfkill')
export _INITRAMFS_WIFI_DRIVERS='brcmfmac brcmutil cfg80211 rfkill bcm43455c0 brcmfmac43455 brcmfmac43430 brcmfmac43430a0' 

# ROOT PASSWORD CHANGER settings ----------------------------------------------
# Hooks
#   optional-sys-rootpassword
#       Changes the system root password

## The new root password
export _ROOTPASSWD="password"

# OPENVPN settings ------------------------------------------------------------
# Hooks
#   optional-sys-openvpn
#       Attemps to configure the system to use an VPN (as a client)

## Zip file containing 'client.conf' and any other deps (creds, certs, etc)
# Optional variable (default: "openvpn.zip")
export _OPENVPN_CONFIG_ZIP="openvpn.zip"
```

loliDe, 2020-12-02:
I have netx error, when Build initframs:
Attempting to build new initramfs ... (CHROOT is /mnt/cryptmypi)
  Creating symbolic links from current physical device to crypttab device (if not using sd card mmcblk0p)
  Using kernel '4.19.118-Re4son-v8l+'
  Building new initramfs ...
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/drivers/md/dm-mod.ko
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/drivers/md/dm-crypt.ko
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/net/rfkill/rfkill.ko
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/net/wireless/cfg80211.ko
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmutil/brcmutil.ko
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4373-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4356-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4354-sdio.bin for module brcmfmac
Adding firmware /lib/firmware/brcm/brcmfmac43455-sdio.bin
Adding firmware /lib/firmware/brcm/brcmfmac43430-sdio.bin
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43430a0-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4339-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43362-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4335-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43341-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43340-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4334-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4330-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4329-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43241b5-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43241b4-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43241b0-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43143-sdio.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac4373.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43569.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43242a.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43236b.bin for module brcmfmac
W: Possible missing firmware /lib/firmware/brcm/brcmfmac43143.bin for module brcmfmac
Copying module directory kernel/drivers/usb/dwc2
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/drivers/usb/gadget/udc/udc-core.ko
Adding module /lib/modules/4.19.118-Re4son-v8l+/kernel/drivers/usb/dwc2/dwc2.ko
Copying module directory kernel/drivers/input/keyboard

unixabg, 2020-12-06:
Greetings,
So the issue was that the firmware for the wifi was not included in the initramfs. Also on boot it needs just a small sleep to allow the wlan0 be ready for operations, see commit 7ecfe01e3 . I tested both rpi3b+ and rpi4 in which both unlock over wlan0. Commit f807abf30 added additional example information to unlock over wifi. Please test and let me know if now working for you.

loliDe, 2020-12-08:
Hi @unixabg 
Thx for your atention, I will test and give to you feedback...
Cheers!

loliDe, 2020-12-10:
@unixabg  IT's A LIVE....
Thx so much for your work Richard, all works ....
Thx!!!!!!!!!!!!!!!!!!!!

closed 2020-12-10 by loliDe