unixabg/cryptmypi #31: Cannot SSH

ftondolo, 2020-12-02:
Have tried ssh'ing into a newly created Dropbear ssh image on an rpi0w but frankly I cannot figure out trough what medium.  I have tried setting it up heedlessly to connect to a Wifi network and even to use g_ether to use a computer's ability to create a bridged network with an RNDIS gadget but the computer refuses to see any device.

unixabg, 2020-12-06:
Greetings ftondolo,
So I do not have the hardware to test on. However wifi unlock does work now on rpi3b+ and rpi4, so with the right config that might now work. Also to make the size a bit smaller and build copy faster, if you do not need full desktop environment (fde), I use:

export _PKGSPURGE="kali-desktop-xfce kali-root-login xserver-xorg-video-fbdev xserver-xorg-input-evdev xserver-xorg-input-synaptics"

As for the wifi settings in the image I used for testing see:  7ecfe01  . So maybe try with the latest from next-4.x and report back.

unixabg, 2020-12-14:
Greetings,
Bumping to see if issue still exists?

unixabg, 2021-02-16:
Greetings,
Final bump, if no response will close soon.

unixabg, 2021-03-09:
Closing since no response on bug.

closed 2021-03-09 by unixabg