unixabg/cryptmypi #32: Dropbear never starts

bigmrw, 2020-12-03:
I have a RPi 4 and am using a pretty basic configuration for cryptmypi.  When booting, dropbear never starts, so I am unable to SSH to unlock the pi.  The last line that I see in the boot log is:

Begin: Starting dropbear ...  <pause of about 1 minute or so before the following>  [  204.893840] random: crng  init done

closed 2020-12-04 by bigmrw

unixabg, 2020-12-05:
Greetings bigmrw,
Would you be willing to share what the solution to your issue was for other users who may encounter a like issue?
Either way thank you for using the project and as always if tagged release doesn't work always try next-4.x branch.

bigmrw, 2020-12-05:
This was my own stupidity - Dropbear was actually starting, but I did not realize it was starting on port 2222.