unixabg/cryptmypi #33: No key available with this passphrase

bigmrw, 2020-12-05:
Hello - I am having some issues getting cryptmypi working properly.  I am using a Raspberry Pi 4B 8GB with a 128GB SD card.

I am using the next-4.x branch of code and using the following hooks in my configuration:

`
stage1_optional_hooks(){
    myhooks "optional-initramfs-luksnuke"
    myhooks "optional-sys-gpugovernor-ondemand"
    myhooks "optional-sys-dns"
    myhooks "optional-sys-docker"
}
`
`
stage2_optional_hooks(){
    myhooks "optional-sys-rootpassword"
    myhooks "optional-sys-vpnclient"
}
`

I am following the process to unlock LUKS via SSH:
`ssh -i /home/user/.ssh/id_rsa -p2222 root@10.11.12.13`

I am then presented with the password prompt to unlock the LUKS partition:
`Enter passphrase for /dev/mmcblk0p2:`

And then get the following error:
`No key available with this passphrase.`

I know the password that I am typing is correct - I have tried both typing manually and copy/paste.  Any help in troubleshooting this would be greatly appreciated.

@unixabg Thank you for your work on this valuable project.

unixabg, 2020-12-06:
Greetings,
I did a test run with examples/kali-encrypted-basic-dropbear and here is the diff from defaults I used:

```

diff --git a/examples/kali-encrypted-basic-dropbear/cryptmypi.conf b/examples/kali-encrypted-basic-dropbear/cryptmypi.conf
index 93a88c7..58ce0a3 100644
--- a/examples/kali-encrypted-basic-dropbear/cryptmypi.conf
+++ b/examples/kali-encrypted-basic-dropbear/cryptmypi.conf
@@ -19,7 +19,7 @@
 #   - Re4son+ is for armv6 devices (ie. RPi1, RPi0, and RPi0w)
 #   - v7+ and v8+ sufixes are for the 32bit and 64bit armv7 devices (ie. RPi 3)
 #   - l+ sufix in the name means they will be ready for the RPi4.
-export _KERNEL_VERSION_FILTER="v8+"
+export _KERNEL_VERSION_FILTER="l+"
 
 # HOSTNAME
 #   Each element of the hostname must be from 1 to 63 characters long and
@@ -78,9 +78,12 @@ stage1_hooks(){
 
 
 #   Optional function: can be ommited.
-#stage1_optional_hooks(){
+stage1_optional_hooks(){
 #    myhooks "experimental-initramfs-wifi"
-#}
+    myhooks "optional-initramfs-luksnuke"
+# myhooks "optional-sys-gpugovernor-ondemand"
+# myhooks "optional-sys-dns"
+}

``` 
And I was able to ssh and unlock. So I would see if you can build this more basic config and unlock. If so then add components one or two at a time to see what is not operating as expected. Please know that some things like the docker hook was a contrib and I do not use or test. If you find which component is causing the issue I am glad to take a look to see if I can assist. And as always patches welcome.

bigmrw, 2020-12-06:
You nailed it.  I was able to get everything working by removing docker from the mix.  This is an awesome project and perfect for my needs, thanks again!

unixabg, 2020-12-06:
Greetings,
I am glad you got it going. If the docker part is important for your setup, you could try just it with the k-e-b-d default and see how it behaves. As I said before there are a few components, docker being one and iodine being another, that I do not test and always appreciate testers. I believe this has resolved your issue and if so please be so kind as to mark closed. Thanks for using the project.

closed 2020-12-07 by bigmrw