unixabg/cryptmypi #34: Any luck booting from a Cryptmypi'ed external USB device?

xmijo, 2020-12-07:
Firstly, I want to say thanks for putting this together, it simplifies an otherwise long and tedious process tremendously.

With Pi's now being able to [boot from USB](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md), it would be great if Cryptmypi had support for USB storage devices, such as an external SSD, as it can provide performance benefits over SD cards (if using USB3.0). SD cards are also prone to failure. To enable USB boot on the Pi, one must first [update the bootloader](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md). [another link](https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb)

I've successfully used Cryptmypi to create an encrypted Kali image on an SD card but haven't yet been able to get Cryptmypi working for USB devices in my testing.

The first attempt with Cryptmypi to create the image on my SSD drive resulted in it not booting at all. I couldn't see if there were any errors showing as I'm going headless, but I figured it was because of the references to _/dev/mmcblk0pX_ devices, which the Pi seems to give to SD cards, whereas a USB device would get _/dev/sdaX_. I updated the reference in boot/initramfs-tools/update.sh in the build dir (before Stage2) and after flashing I mounted the partitions and edited the references in _cmdline.txt, fstab, and crypttab._

I also downloaded the latest *.dat and *.elf files from the [Raspberry Pi firmware repo](https://github.com/raspberrypi/firmware/tree/master/boot) and copied them to the SSD boot partition. I suppose the Kali ARM image doesn't have the latest versions of those yet.

Next attempt was more successful. Trying to SSH in, I reach the LUKS prompt that asks me to unlock my device and enter the password, and the connection closes. But when SSH'ing back in again it immediately drops the connection. This tells me that the boot is successful but it hangs after providing the LUKS password.

Just want to see if anyone has gotten this to work or can see if I've missed anything that I should test?

unixabg, 2021-02-16:
Greetings,
I have held off on replying on this issue to see what interest or comments there might be on this feature/ability. Please know cryptmypi 4.6-beta was just released and luks now unlocks with uuid instead of hard coded block device. So, maybe this help on this issue. If you would be so kind as to make an attempt with the latest and see if this moves this issue forward it would be appreciated.

xmijo, 2021-02-18:
Hi, thanks for the reply! I will try it with the new version soon and report back.

unixabg, 2021-03-09:
Greetings,
Did you have any luck on this?