unixabg/cryptmypi #36: DropBear Unlock via LTE\3G USB modem!

loliDe, 2020-12-15:
Hi @unixabg 
How I can unlock my encrypted kali-complete Rpi4B via LTE\3G USB modem (eg. Huawei E3372h-153)?

IF its possible to load this USB modem to initframs?
Thx?

unixabg, 2020-12-15:
Greetings,
In general, if you can boot the Rpi normal and get the modem working then yes, it should be possible. You will need the correct modules, binary/firmware components, and interface name and build in the initramfs. It should be very similar to how the *-wifi.hook files work. I do not have like hardware to test, but as always patches welcome.

loliDe, 2020-12-16:
@unixabg  Thx
Go to Work with it =)

loliDe, 2020-12-27:
Hi @unixabg 
Can you help me with hooks and modules, binary/firmware with my issue? Write plz some examples and I will use it for my work..
Cheers and Merry Christmas!

loliDe, 2021-02-09:
Hi @unixabg  I still here
Can you help me?
How I can find correct modules, binary/firmware with my issue?
Thx

unixabg, 2021-02-16:
Greetings,
I do not have hardware to test with so there is little I can do to help. I can leave open for a while longer, but this seems to be more of a specialty request or custom out of mainline request. The only thing I can offer is if you would want to sponsor an attempted solution to contrib to the the project. The contribs are one way, means sponsor pays for all costs up front. Hardware and services I procure for attempted solution is non refundable, no equipment returns to sponsor, and no guarantees I can make work as you want. In a nutshell all costs on you and no returns or refunds. If this is something you wish to pursue for you and other users of the project please email me with this issue # and header in subject line and we work through specifics.

iTosun, 2021-06-09:
I've got this working correctly on my setup. You should add the following code in one of your stages (i used stage 1)
```
# USB mass storage
printf "xhci_hcd\n" >> /etc/initramfs-tools/modules
# USB modem for LTE
printf "cdc_ether\n" >> /etc/initramfs-tools/modules
```
used USB LTE modem: huawei e3372 lte (https://www.amazon.com/Huawei-E3372-Surfstick-microSD-wei%C3%A1/dp/B0104LV06M)

unixabg, 2025-12-03:
Greetings,
Since I am unable to test this and post states "got this working" I am going to close.

closed 2025-12-03 by unixabg