unixabg/cryptmypi #37: Device /dev/mmcblk0p2 doesn't exist or access denied.

donnm, 2021-01-09:
I'm trying to use examples/pios-encrypted-basic-dropbear on a Raspberry Pi 4 Model B 8GB and it boots and accepts the SSH connect but the following messages show up instead of the LUKS password prompt:

```
Device /dev/mmcblk0p2 doesn't exist or access denied.
Device /dev/mmcblk0p2 doesn't exist or access denied.
Device /dev/mmcblk0p2 doesn't exist or access denied.
Device /dev/mmcblk0p2 doesn't exist or access denied.
...
```

The image in question is https://raspi.debian.net/verified/20201112_raspi_4.img.xz. I tracked it down to the generated initramfs:

```
etc/unlock.sh:7:    test -e /dev/mapper/crypt && break || cryptsetup luksOpen /dev/mmcblk0p2 crypt
```

which comes from the encryption hook:

```
3000-stage1-setup-encryption.hook:67:    test -e /dev/mapper/crypt && break || cryptsetup luksOpen /dev/mmcblk0p2 crypt
```

In the boot messages I see that the SD card is detected as mmcblk1, so this explains the error. Is this particular to the Pi 4?

I am able to get it working by updating the hook to refer to mmcblk1p2 and by modifying /etc/fstab, /etc/crypttab, and /boot/cmdline.txt because the patterns in those files are not matched by the sed commands looking for /dev/mmcblk0p2. This is caused by references to filesystem labels:

This line in /etc/fstab
```
LABEL=RASPIROOT / ext4 rw 0 1
```
becomes:
```
/dev/mapper/crypt / ext4 rw 0 1
```

This line in /etc/crypttab
```
crypt    /dev/mmcblk0p2    none    luks
```
becomes:
```
crypt    /dev/mmcblk1p2    none    luks
```

And /boot/cmdline.txt
```
console=tty0 console=ttyS1,115200 root=LABEL=RASPIROOT rw elevator=deadline fsck.repair=yes net.ifnames=0 rootwait
```
becomes:
```
console=tty0 console=ttyS1,115200 root=/dev/mapper/crypt cryptdevice=/dev/mmcblk1p2:crypt rootfstype=ext4 rw elevator=deadline fsck.repair=yes net.ifnames=0 rootwait
```

I'm submitting this issue so that others can use this workaround.

unixabg, 2021-02-09:
Greetings,
First thank you for the report and information you have provided. I will flag as enhancement since hard coded target is not ideal. One thing just for clarity, if I could ask you to also submit your /boot/config.txt file on the finished image.

diederikdehaas, 2021-02-13:
Possibly related? https://salsa.debian.org/raspi-team/image-specs/-/issues/31

commit 40871dba (2021-02-13): Closes #37 . Initial commit to move luks unlock to uuid
in hooks/3000-stage1-setup-encryption.hook and hooks/5200-stage2-setup-luks-create.hook.
closed 2021-02-14 by unixabg