unixabg/cryptmypi #38: Chroot Error During Image Build - Failed To Run Command 'apt-get'

bruestle2, 2021-02-01:
OS: Linux Mint (fully updated)
Command Ran: sudo ./cryptmypi.sh examples/pios-encrypted-basic
Error: chroot: failed to run command ‘apt-get’: No such file or directory

Full Log:
main@htpc:~/cryptmypi$ sudo ./cryptmypi.sh examples/pios-encrypted-basic

###############################################################################
                               C R Y P T M Y P I
###############################################################################

Loading functions...
./cryptmypi.sh: line 257: [: /home/main/cryptmypi/functions/chroot: binary operator expected
- Loading chroot.fns ...
  ... chroot.fns loaded!
- Loading echo.fns ...
  ... echo.fns loaded!
- Loading files.fns ...
  ... files.fns loaded!
- Loading hooks.fns ...
  ... hooks.fns loaded!
- Loading misc.fns ...
  ... misc.fns loaded!
- Loading ssh.fns ...
  ... ssh.fns loaded!
- Loading stage1profiles.fns ...
  ... stage1profiles.fns loaded!

Attempting to run preconditions hooks ...
- Calling 0000-preconditions-root.hook ...
    ... 0000-preconditions-root.hook completed!

- Calling 0200-preconditions-rsync.hook ...
    ... 0200-preconditions-rsync.hook completed!

- Calling 0300-preconditions-dns.hook ...
Primary DNS server _DNS1 is not set on config: Setting default value 1.1.1.1
Secondary DNS server _DNS2 is not set on config: Setting default value 8.8.8.8
    ... 0300-preconditions-dns.hook completed!

# Executing both stages #######################################################
###############################################################################
                               C R Y P T M Y P I
                               ---- Stage 1 ----
v4.5-beta
###############################################################################

--- Custom STAGE1 SELECTED
    stage1profile_encryption

--- Executing:

Attempting to run stage1-sanity hooks ...
- Calling 1100-stage1-sanity-qemu.hook ...
    ... 1100-stage1-sanity-qemu.hook completed!

Attempting to run stage1-image-download hooks ...
- Calling 2000-stage1-image-download.hook ...
ARM image 2020-08-20-raspios-buster-armhf-lite.zip already exists!
Validating image checksum...
Done: Image is valid!
    ... 2000-stage1-image-download.hook completed!

Attempting to run stage1-image-extract hooks ...
- Calling 2100-stage1-image-extract.hook ...
Extracting image: 2020-08-20-raspios-buster-armhf-lite.zip ...
Mounting loopback ...
Syncing image/root to /home/main/cryptmypi/examples/pios-encrypted-basic/build/root ...
Syncing image/root/boot to /home/main/cryptmypi/examples/pios-encrypted-basic/build/root/boot ...
Cleaning loopback ...
    ... 2100-stage1-image-extract.hook completed!

Attempting to run stage1-setup-chroot hooks ...
- Calling 2200-stage1-setup-chroot.hook ...
- Copying qemu emulator to chroot ...
Preparing RPi chroot mount structure at '/home/main/cryptmypi/examples/pios-encrypted-basic/build/root'.
Mounting '/home/main/cryptmypi/examples/pios-encrypted-basic/build/root/dev/' ...
Mounting '/home/main/cryptmypi/examples/pios-encrypted-basic/build/root/dev/pts' ...
Mounting '/home/main/cryptmypi/examples/pios-encrypted-basic/build/root/sys/' ...
Mounting '/home/main/cryptmypi/examples/pios-encrypted-basic/build/root/proc/' ...
Fixing ld.so.preload
/home/main/cryptmypi/examples/pios-encrypted-basic/build/root/etc/resolv.conf exists and appears to be not empty.
Updating apt-get
chroot: failed to run command ‘apt-get’: No such file or directory
    ... 2200-stage1-setup-chroot.hook completed!

Attempting to run stage1-locale hooks ...
- Calling 2300-stage1-locale.hook ...
Configuring system locale...
_LOCALE not set, using default value 'en_US.UTF-8'
Uncommenting locale 'en_US.UTF-8' for inclusion in generation
Updating /etc/default/locale
Installing locales
- Installing locales
chroot: failed to run command ‘apt-get’: No such file or directory
apt-get failed: Trying to recover...
chroot: failed to run command ‘/bin/bash’: No such file or directory
ERROR: Could not install locales correctly... Exiting.
Tearing down RPi chroot mount structure at '/home/main/cryptmypi/examples/pios-encrypted-basic/build/root'.
Reverting ld.so.preload fix
Unmounting binds
umount: /dev/sdb: not mounted.
umount: /mnt/cryptmypi: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /dev/mapper/crypt: no mount point specified.
Device crypt is not active.

unixabg, 2021-02-01:
Greetings,
Could you try a run with the next-4.x branch and report back with results. ty

bruestle2, 2021-02-02:
I tried with next-4.x and had the same result. However, I started worrying this might be a PEBKAC error.

I double checked which version of Linux Mint I was running and found I was a major version behind. (When I said above I was updated, I was just running apt update/upgrade). I'm now on Mint 20.1 and the script is working well.

Sorry for the trouble and thank you for making this great tool!

closed 2021-02-02 by bruestle2