unixabg/cryptmypi #39: failed to connect to non-global ctrl_ifname

RustyPixel, 2021-02-05:
Hi great work and thanks.

I have been trying for a few days now to get a new encrypted system up and running from a already known working non encrypted one.

The problem is under `experimental-initramfs-wifi` the wifi driver loads but i never comes up. 

Below is the following errors i get with putty when i follow via tty 
```
[    7.268685] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[    7.411153] usb 1-1.2: New USB device found, idVendor=07d1, idProduct=3303, bcdDevice= 2.00
[    7.128655] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[    7.271137] usb 1-1.2: New USB device found, idVendor=07d1, idProduct=3303, bcdDevice= 2.00
[    7.286048] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    7.296796] usb 1-1.2: Product: 11n Adapter
[    7.304363] usb 1-1.2: Manufacturer: Manufacturer Realtek
[    7.313270] usb 1-1.2: SerialNumber: 00e04c000001
[   18.408087] Freeing unused kernel memory: 1024K
[   18.416401] Run /init as init process
[   19.739581] r8712u: module is from the staging directory, the quality is unknown, you have been warned.
[   19.752709] smsc95xx 1-1.1:1.0 enxb827eb18d407: renamed from eth0
[   19.786472] r8712u: register rtl8712_netdev_ops to netdev_ops
[   19.795871] usb 1-1.2: r8712u: USB_SPEED_HIGH with 4 endpoints
[   19.810473] usb 1-1.2: r8712u: Boot from EFUSE: Autoload OK
[   20.314131] random: crng init done
[   20.766779] usb 1-1.2: r8712u: CustomerID = 0x0000
[   20.774589] usb 1-1.2: r8712u: MAC Address from efuse = 28:10:7b:b6:d4:ac
[   20.784337] usb 1-1.2: r8712u: Loading firmware from "rtlwifi/rtl8712u.bin"
[   20.796091] usbcore: registered new interface driver r8712u
[   20.816163] r8712u 1-1.2:1.0 wlx28107bb6d4ac: renamed from wlan0
[   20.919333] device-mapper: ioctl: 4.41.0-ioctl (2019-09-16) initialised: dm-devel@redhat.com
[   21.026821] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   21.122972] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   21.133118] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[   21.151337] cfg80211: failed to load regulatory.db
[   21.204630] usbcore: registered new interface driver brcmfmac
```

please also see screenshot of monitor output

[image]

Here is also my config file
```
export _KERNEL_VERSION_FILTER="v7+"

export _HOSTNAME="encrypted-pi"

export _LOCALE="en_GB.UTF-8"

export _BLKDEV="/dev/sdb"

export _LUKSCIPHER="aes-cbc-essiv:sha256"

export _LUKSPASSWD="berry"

export _LUKSEXTRA=""

export _IMAGEURL=https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/2020-08-20-raspios-buster-armhf-lite.zip
export _IMAGESHA="4522df4a29f9aac4b0166fbfee9f599dab55a997c855702bfe35329c13334668"

export _PKGSPURGE=""
export _PKGSINSTALL=""

export _SSH_LOCAL_KEYFILE="$_USER_HOME/.ssh/id_rsa"

stage1_hooks(){
    stage1profile_complete
}

stage1_optional_hooks(){
    myhooks "experimental-initramfs-wifi"
}

stage2_optional_hooks(){
    myhooks "optional-sys-rootpassword"
}

export _DNS1='1.1.1.1'

export _DNS2='1.0.0.1'

export _WIFI_SSID='testwifi'

export _WIFI_PASS='testwifipassword'

export _WIFI_INTERFACE='wlan0'

export _INITRAMFS_WIFI_INTERFACE='wlan0'

export _INITRAMFS_WIFI_IP=":::::${_INITRAMFS_WIFI_INTERFACE}:dhcp:${_DNS1}:${_DNS2}"

### ** NOTE r8712u is a driver already in raspios-buster-armhf-lite **
export _INITRAMFS_WIFI_DRIVERS='brcmfmac brcmutil cfg80211 rfkill r8712u'

export _ROOTPASSWD="raspberry"
```

I have also been able to compile in a a new wifi module  `8821cu` that also loads but i get the same error.

Unit: Raspberry pi 2
OS: as in config file
WIFI: wifi card DWA-131 `r8712u`
WIFI: wifi card cudy WU650 `8821cu`

any advise would be great.
Thanks

unixabg, 2021-02-07:
Greetings,
Not really sure, I at this time, only test on rpi3b+ and rpi4. I did do a sample run on next-4.x branch with:

rpi3b+ and examples/kali-encrypted-basic-dropbear and all wifi settings set

Run was successful and luks unlocked over wifi. You might try the examples/kali-encrypted-basic-dropbear and see if works for you. Please verify the image from kali you need for your hardware.

RustyPixel, 2021-02-12:
> I at this time, only test on rpi3b+ and rpi4

These have onboard WiFi and which limits the type of case i want to used at a distance, there is no IPEX connector for external antennas

I have got it  working partly with using your `examples/pios-encrypted-basic-dropbear` as the base for the install and then setup my WiFi manually.

I then look looked at the `hooks/0000-experimental-initramfs-wifi.hook` and followed the link to the references notes,    http://www.marcfargas.com/posts/enable-wireless-debian-initramfs/ which has a  git gist page.

There i used the scripts in my tests and found that i was still getting the same errors and complaints about wlan0. 

I decide change  `INTERFACE="wlan0"`  to  `INTERFACE="wlx28107bb6d4ac"`  ( mac address of the wifi card ) and also 

`/sbin/wpa\_supplicant -i$INTERFACE -c/etc/wpa\_supplicant.conf -P/run/initram-wpa\_supplicant.pid -B -f /tmp/wpa\_supplicant.log`

to

`/sbin/wpa\_supplicant -Dnl80211,wext -i$INTERFACE -c/etc/wpa\_supplicant.conf -P/run/initram-wpa\_supplicant.pid -B -f /tmp/wpa\_supplicant.log`

i added  `ip=:::::wlx28107bb6d4ac:dhcp:1.1.1.1:1.0.0.1` to  /boot/cmdline.txt

Now the WiFi came up but I still  having ifconfig complained about too many parameters, so i change above to match

`ip=:::::wlx28107bb6d4ac:on`.

Now it all working  and pulling a it own ip address, dns still gives me errors when i add.

RustyPixel, 2021-02-12:
Predictable-Network-Interface-Names was the problem for me with wlan0

Fixed by creating */etc/initramfs-tools/10-rename-network.rules* with

`SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="28:10:7b:b6:d4:ac", NAME="wlan0"`

then in */etc/initramfs-tools/hooks/enable-wireless* added

`copy_file config /etc/initramfs-tools/10-rename-network.rules /usr/lib/udev/rules.d/10-rename-network.rules`

This problem has now been fixed for my but i hope this information helps other stuck with the same or similar problem

closed 2021-02-12 by RustyPixel