unixabg/cryptmypi #41: Fail to build - examples/debian-encrypted-basic-dropbear

RustyPixel, 2021-02-14:
Failed to build also notice the `update-initramfs: Generating /boot/initrd.img-5.10.0-3-arm64` line

`256 SHA256:MPXGEWVQXPfZODBuXpoKZEtkwFk8ptPzsDIWrfnojEs /etc/dropbear-initramfs/dropbear_ed25519_host_key (ED25519)
+--[ED25519 256]--+
|     ..== +**.. .|
|      o++o +.o o+|
|      o=+.+ o +.o|
|      +*=o o + . |
|       =S=  +    |
|      * ....     |
|    E. =  .      |
|   . o. .        |
|    ooo          |
+----[SHA256]-----+
update-initramfs: deferring update (trigger activated)
Dropbear has been added to the initramfs. Don't forget to check
your "ip=" kernel bootparameter to match your desired initramfs
ip configuration.

Processing triggers for libc-bin (2.31-9) ...
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
dpkg: error processing package libc-bin (--configure):
 installed libc-bin package post-installation script subprocess returned error exit status 139
Processing triggers for initramfs-tools (0.139) ...
update-initramfs: Generating /boot/initrd.img-5.10.0-3-arm64
cryptsetup: WARNING: Couldn't determine root device
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
dropbear: WARNING: Invalid authorized_keys file, SSH login to initramfs won't work!
W: Couldn't identify type of root file system for fsck hook
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
/usr/bin/grep: /proc/device-tree/model: No such file or directory
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)
apt-get failed: Trying to recover...
+ sleep 5
+ apt-get update
Hit:1 http://deb.debian.org/debian bullseye InRelease
Reading package lists... Done
+ apt-get -y install dropbear
Reading package lists... Done
Building dependency tree
Reading state information... Done
dropbear is already the newest version (2020.81-3).
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "en_GB:en",
        LC_ALL = (unset),
        LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory
Setting up libc-bin (2.31-9) ...
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
dpkg: error processing package libc-bin (--configure):
 installed libc-bin package post-installation script subprocess returned error exit status 139
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)
+ exit 1
ERROR: Could not install dropbear correctly... Exiting.
Tearing down RPi chroot mount structure at '/home/notroot/cryptmypi/examples/debian-encrypted-basic-dropbear/build/root'.
Unmounting binds
umount: /dev/sdb*: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /dev/mapper/crypt: no mount point specified.
Device crypt is not active.
`

unixabg, 2021-02-14:
Greetings,
I just released 4.6-beta today both master and next-4.x branches should build that image.

RustyPixel, 2021-02-14:
I am using the 4.6-beta on Debian

```Distributor ID: Debian
Description:    Debian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
Linux debian 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux
```

I just tried to build pi2 image with this error

```- Calling 4800-stage1-initramfs.hook ...
Attempting to build new initramfs ... (CHROOT is /home/pi/debian-encrypted-basic-dropbear/build/root)
  Creating symbolic links from current physical device to crypttab device (if not using sd card mmcblk0p)
  Using kernel ''
  Building new initramfs ...
W: missing /lib/modules/4.19.0-14-amd64
W: Ensure all necessary drivers are built into the linux image!
depmod: ERROR: could not open directory /lib/modules/4.19.0-14-amd64: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
/usr/bin/cat: /var/tmp/mkinitramfs_ylW7SL/lib/modules/4.19.0-14-amd64/modules.builtin: No such file or directory
/usr/bin/find: '/var/tmp/mkinitramfs_ylW7SL/lib/modules/4.19.0-14-amd64/kernel': No such file or directory
```

and

``` Adding config /etc/initramfs-tools/unlock.sh
/usr/share/initramfs-tools/scripts/local-bottom/ORDER ignored: not executable
/usr/share/initramfs-tools/scripts/init-bottom/ORDER ignored: not executable
/usr/share/initramfs-tools/scripts/local-top/ORDER ignored: not executable
/usr/share/initramfs-tools/scripts/local-premount/ORDER ignored: not executable
/usr/share/initramfs-tools/scripts/local-block/ORDER ignored: not executable
/usr/share/initramfs-tools/scripts/init-top/ORDER ignored: not executable
/usr/share/initramfs-tools/scripts/init-premount/ORDER ignored: not executable
depmod: WARNING: could not open modules.order at /var/tmp/mkinitramfs_ylW7SL/lib/modules/4.19.0-14-amd64: No such file or directory
depmod: WARNING: could not open modules.builtin at /var/tmp/mkinitramfs_ylW7SL/lib/modules/4.19.0-14-amd64: No such file or directory
Building cpio /boot/initramfs.gz initramfs
  Cleaning up symbolic links
    ... 4800-stage1-initramfs.hook completed!

- Calling 4900-stage1-teardown-chroot.hook ...
Tearing down RPi chroot mount structure at '/home/pi/debian-encrypted-basic-dropbear/build/root'.
Unmounting binds
    ... 4900-stage1-teardown-chroot.hook completed!

###############################################################################
                               C R Y P T M Y P I
                               ---- Stage 2 ----
v4.6-beta
############################################################################### 
```

note the warnings and referances to amd64 which should not be happening.

unixabg, 2021-02-15:
Greetings,
So thank you for the extra details. On Debian 10 (Buster) I can reproduce the issue. Not exactly sure why, but I will dig around a bit and see what I can discover. In the short term I would ask that you please consider spinning up a VM of Kali, testing, and reporting back if that is in working order for you. I will leave issue open until I can see why this behavior is occurring on Debian 10.

unixabg, 2021-02-15:
Greetings,
I think it is something with the version of qemu on Debian. To get a later version of qemu I have added backports to my sources.list:

`deb http://deb.debian.org/debian/ buster-backports main contrib non-free
`

then 

`apt update && apt install qemu-user-static/buster-backports`

and try another run.

RustyPixel, 2021-02-16:
I change over to Kali and was able to build Pi 2 buster and bullseye.

Both seems to have problems in `- Calling 2500-stage1-packages.hook ... ` in relation to `/usr/bin/locale: Cannot set ` 

Pi2 Buster build has this error

`Setting up dropbear (2018.76-5) ...
Processing triggers for systemd (241-7~deb10u6) ...
Processing triggers for libc-bin (2.28-10) ...
Processing triggers for initramfs-tools (0.133+deb10u1) ...
update-initramfs: Generating /boot/initrd.img-4.19.0-14-armmp
cryptsetup: WARNING: Couldn't determine root device
cryptsetup: WARNING: The initramfs image may not contain cryptsetup binaries
    nor crypto modules. If that's on purpose, you may want to uninstall the
    'cryptsetup-initramfs' package in order to disable the cryptsetup initramfs
    integration and avoid this warning.
dropbear: WARNING: Invalid authorized_keys file, remote unlocking of cryptroot via SSH won't work!
W: Couldn't identify type of root file system for fsck hook
- Installing dropbear-initramfs
`

pi2  _KERNEL_VERSION_FILTER is **armmp**
pi3 and 4 _KERNEL_VERSION_FILTER is **arm64**

in stage2-otherscript.sh i change 
__DEBIAN_KERNEL="initrd.img-x.x.x-x-ARCH" 
to
`__DEBIAN_KERNEL="initrd.img-$(ls /usr/lib | grep linux-image | cut -d\- -f3,4,5)"`

doing some test.

closed 2021-02-23 by RustyPixel