unixabg/cryptmypi #42: AES-cbc vs AES-xts performance on the RPi

xmijo, 2021-02-20:
Hi,
Regarding the cryptsetup encryption ciphers, is there a good reason why "aes-cbc-essiv:sha256" is the default in the example files when "aes-xts-plain64" appears to have better performance on Raspberry Pis? I know the cipher can be changed in the example files, of course, but just interested in hearing if there's any reason why it is used as default.

`cryptsetup benchmark` shows:
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b        23.8 MiB/s        77.7 MiB/s
    serpent-cbc        128b               N/A               N/A
    twofish-cbc        128b        55.8 MiB/s        56.2 MiB/s
        aes-cbc        256b        17.4 MiB/s        58.9 MiB/s
    serpent-cbc        256b               N/A               N/A
    twofish-cbc        256b        55.8 MiB/s        56.1 MiB/s
        aes-xts        256b        85.0 MiB/s        74.9 MiB/s
    serpent-xts        256b               N/A               N/A
    twofish-xts        256b        61.1 MiB/s        60.4 MiB/s
        aes-xts        512b        65.4 MiB/s        57.4 MiB/s
    serpent-xts        512b               N/A               N/A
    twofish-xts        512b        61.3 MiB/s        60.3 MiB/s

unixabg, 2021-02-20:
Greetings,
First thank you for the information. As to the reason why  "aes-cbc-essiv:sha256" is used, it was used in the example I build the scripts to automate:  https://www.kali.org/docs/arm/raspberry-pi-full-encryption/  . I do believe "aes-cbc-essiv:sha256" provides sufficient security, but it would appear not best in performance. I could make a note in the example configs of other options and potential performance improvements OR just leave as is and know if users encounter performance issues they could dig deep enough to find this issue and information.

thomas725, 2022-10-13:
From https://rr-developer.github.io/LUKS-on-Raspberry-Pi/ I've found that `xchacha20,aes-adiantum-plain64` has a much better performance still, but it seem's for that we'd need to copy additional kernel modules into initramfs. 
```
root@pios-encrypted-basic-dropbear-arm64:~# cryptsetup benchmark -c xchacha20,aes-adiantum-plain64
# Tests are approximate using memory only (no storage IO).
#            Algorithm |       Key |      Encryption |      Decryption
xchacha20,aes-adiantum        256b       136.7 MiB/s       140.8 MiB/s
```

unixabg, 2025-12-03:
Greetings,

Since the value can be changed users are free to do so as they wish in the configs.