unixabg/cryptmypi #43: apt update fails on debian

librecoder42, 2021-04-07:
Hello,

when I try to run ./cryptmypi.sh examples/debian-encrypted-basic-dropbear/ it fails at this point:

```
Calling 2300-stage1-locale.hook ..
...
+ apt-get -y install locales
...
E: Package 'locales' has no installation candidate
+ exit 1
ERROR: Could not install locales correctly... Exiting.
```

But I think that the **root error is that apt update fails before that because /etc/resolv.conf has the wrong permissions**:

```
- Calling 2200-stage1-setup-chroot.hook ...
...
/tmp/cryptmypi/examples/debian-encrypted-basic-dropbear//build/root/etc/resolv.conf does not exists or is empty.
Setting nameserver to 1.1.1.1 and 8.8.8.8 in /tmp/cryptmypi/examples/debian-encrypted-basic-dropbear//build/root/etc/resolv.conf
Updating apt-get
Err:1 http://deb.debian.org/debian bullseye InRelease
  Temporary failure resolving 'deb.debian.org'
Reading package lists... Done
W: Failed to fetch http://deb.debian.org/debian/dists/bullseye/InRelease  Temporary failure resolving 'deb.debian.org'
```

When I look in the RPi chroot I find this:
```
ls -lah /etc/resolv.conf 
-rw------- 1 root root 38 2021-04-07 19:45 /etc/resolv.conf
```

And inside the chroot "apt update" fails.

When I change /etc/resolv.conf to the imho correct 644 permission "apt update" works inside the chroot.

It also works again if I add this line in functions/chroot.fns (after the resolv.conf is created)
`chmod 644 "${CHROOTDIR}/etc/resolv.conf"`

Btw, thanks for this great project!

unixabg, 2021-04-16:
Greetings,
Can you share with me what version you are running? I think this is an old bug that has since been addressed. Version in master branch is 4.7-beta and next-4.x is 4.8-next,

librecoder42, 2021-04-16:
I am commit 64c2a4bff958c8d9f2b7ecd078952b964d4d4dff (HEAD -> master, tag: v4.7-beta, origin/master, origin/HEAD)

unixabg, 2021-04-17:
Greetings,
Interesting, can you share your distro and build environment?
Also can you try next-4.x branch to see if it behaves the same?
I tested for the issue on a latest pristine Kali vm with next-4.x branch and could not reproduce the issue.

librecoder42, 2021-04-17:
I am running Debian sid.
Linux silverweed 5.10.0-2-amd64 #1 SMP Debian 5.10.9-1 (2021-01-20) x86_64 GNU/Linux
GNU bash, version 5.1.4(1)-release (x86_64-pc-linux-gnu)

What would you define as my build environment? I am happy to give more information.
Can you reproduce this with the debian-encrypted-basic-dropbear  example?

unixabg, 2021-04-17:
Greetings,
Your info sufficient on build environment. Also I work to support Debian so all good to report issue. I run tests in pristine vm's so I try to avoid missing simple things. I tested next-4.x branch on pristine Debian Bullseye vm and could not reproduce issue. If next-4.x not work for you I build a sid vm and try to reproduce. Just let me know.

librecoder42, 2021-04-19:
The error is still there:

```
/tmp/newcrypt/cryptmypi# git log | head -1 && git branch
commit e153f05cdd68e4751d75e02ce5f0059b0112e481
  master
* next-4.x
```
```
./cryptmypi.sh examples/debian-encrypted-basic-dropbear/

###############################################################################
                               C R Y P T M Y P I
###############################################################################

Loading functions...
- Loading chroot.fns ...
  ... chroot.fns loaded!
- Loading echo.fns ...
  ... echo.fns loaded!
- Loading files.fns ...
  ... files.fns loaded!
- Loading hooks.fns ...
  ... hooks.fns loaded!
- Loading misc.fns ...
  ... misc.fns loaded!
- Loading ssh.fns ...
  ... ssh.fns loaded!
- Loading stage1profiles.fns ...
  ... stage1profiles.fns loaded!

Attempting to run preconditions hooks ...
- Calling 0000-preconditions-root.hook ...
    ... 0000-preconditions-root.hook completed!

- Calling 0200-preconditions-rsync.hook ...
    ... 0200-preconditions-rsync.hook completed!

- Calling 0300-preconditions-dns.hook ...
Primary DNS server _DNS1 is not set on config: Setting default value 1.1.1.1
Secondary DNS server _DNS2 is not set on config: Setting default value 8.8.8.8
    ... 0300-preconditions-dns.hook completed!

- Calling 0400-preconditions-cryptsetup.hook ...
    ... 0400-preconditions-cryptsetup.hook completed!

# Executing both stages #######################################################
###############################################################################
                               C R Y P T M Y P I
                               ---- Stage 1 ----
v4.8-next
###############################################################################

--- Custom STAGE1 SELECTED
    stage1profile_complete

--- Executing:

Attempting to run stage1 hooks ...
- Calling 1100-stage1-sanity-qemu.hook ...
    ... 1100-stage1-sanity-qemu.hook completed!

- Calling 2000-stage1-image-download.hook ...
Downloading ARM image from https://raspi.debian.net/verified/20210408_raspi_4_bullseye.img.xz ...
--2021-04-19 11:26:52--  https://raspi.debian.net/verified/20210408_raspi_4_bullseye.img.xz
Resolving raspi.debian.net (raspi.debian.net)... 208.97.148.173
Connecting to raspi.debian.net (raspi.debian.net)|208.97.148.173|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 353290740 (337M) [application/x-xz]
Saving to: ‘/tmp/newcrypt/cryptmypi/files/images/20210408_raspi_4_bullseye.img.xz’
[... 64 lines cut ...]
is only available from another source

E: Package 'locales' has no installation candidate
+ exit 1
ERROR: Could not install locales correctly... Exiting.
Tearing down RPi chroot mount structure at '/tmp/newcrypt/cryptmypi/examples/debian-encrypted-basic-dropbear//build/root'.
Unmounting binds
umount: /dev/sdb*: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /mnt/cryptmypi: no mount point specified.
umount: /dev/mapper/crypt: no mount point specified.
Device crypt is not active.
```
```
/tmp/newcrypt/cryptmypi# chroot /tmp/newcrypt/cryptmypi/examples/debian-encrypted-basic-dropbear//build/root
root@silverweed:/# stat /etc/resolv.conf 
  File: /etc/resolv.conf
  Size: 38              Blocks: 8          IO Block: 4096   regular file
Device: fd01h/64769d    Inode: 24670733    Links: 1
Access: (0600/-rw-------)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2021-04-19 09:28:14.130947096 +0000
Modify: 2021-04-19 09:28:14.130947096 +0000
Change: 2021-04-19 09:28:14.130947096 +0000
 Birth: 2021-04-19 09:28:14.130947096 +0000
```

commit 80229373 (2021-04-28): Closes #43. Force access permission on chroot /etc/resolv.conf to 644.
closed 2021-06-09 by unixabg