unixabg/cryptmypi #47: Dropbear SSH Server fails on current RPi Builds

bigmrw, 2021-09-03:
I just installed cryptmypi on a Raspberry Pi 4 using kali-linux-2021.2-rpi4-nexmon-64.img.xz and found that in order to get dropbear SSH server running in initramfs I had to updated the dropbear hook (2700-stage1-dropbear.hook) as follows:
`Replace /etc/dropbear-initramfs/config with /etc/dropbear/initramfs/config`

bigmrw, 2021-09-05:
In the end, the change was required in more than just this file.  From the root of clean cryptmypi source, you can run the following command to update all source files:
egrep -lRZ '/etc/dropbear-initramfs/' . | xargs -0 -l sed -i -e 's;/etc/dropbear-initramfs/;/etc/dropbear/initramfs/;g'

unixabg, 2021-09-26:
Greetings,
Issue confirmed. I did a bit of looking and found in Debian where the change was introduced:

https://salsa.debian.org/debian/dropbear/-/commit/191ded3f2d139586c0473f724c0a325ca209ead5

I will work on a way to maintain backwards compatibility and accept what appears to be the future behavior in dropbear.

commit ca4b30cd (2021-09-28): Closes #47. Maintain dropbear compatibility with latest and older images in hooks/2700-stage1-dropbear.hook.
closed 2021-10-01 by unixabg