unixabg/cryptmypi #49: Device  /dev/disk/by-uuid/XXX...XXXX doesn't exist or access denied

donyshmony, 2022-02-07:
**Hi there @unixabg** 
First of all thx for the GREAT work with this project, **BUT** now I have some Big problem and I am stuck on these:
I'm trying to use **examples/pios-encrypted-basic-dropbear** on a _Raspberry Pi (4 Model B Rev 1.2 4 GB Ram)_:
```
rooy@pi-home:/home# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03112
Serial          : 10000000f4031d51
Model           : Raspberry Pi 4 Model B Rev 1.2
```

 and it boots and accepts the SSH connect but the following messages show up instead of the LUKS password prompt:
```
Device /dev/disk/by-uuid/609fe123-f56d-4125-ba8d-986a77033580 does not exist or access denied.
Device /dev/disk/by-uuid/609fe123-f56d-4125-ba8d-986a77033580 does not exist or access denied.
Device /dev/disk/by-uuid/609fe123-f56d-4125-ba8d-986a77033580 does not exist or access denied.
```
And that's all I can't log in and unlock my RPi
 I am using the next config files (I have  just updated img url and sha256, and change KERNEL_VERSION_FILTER to v8l+ ):

```
###############################################################################
## cryptmypi profile ##########################################################

# EXAMPLE OF A SIMPLE ENCRYPTED Raspberry Pi OS CONFIGURATION
#   Will create a encrypted pios system:
#   - during boot the encryption password will be prompted both
#       via the console and ssh on port 2222
#   - with ssh server (available after boot)
#       The id_rsa.pub public key will be added to authorized_keys
#
#   Some optional hooks are defined on stage2:
#   - "optional-sys-rootpassword" that sets root password

# General settings ------------------------------------------------------------
# You need to choose a kernel compatible with your RPi version.
#   - Re4son+ is for armv6 devices (ie. RPi1, RPi0, and RPi0w)
#   - v7+ and v8+ are for the 32bit and 64bit armv7 devices (ie. RPi 3)
#   - l+ is mostly for the RPi4 since one can have 4GB or 8GB versions
#     The l in l+ actually means lpae (Large Physical Address Extensions)
export _KERNEL_VERSION_FILTER="v8l+"

# HOSTNAME
#   Each element of the hostname must be from 1 to 63 characters long and
#   the entire hostname, including the dots, can be at most 253
#   characters long.  Valid characters for hostnames are ASCII(7) letters
#   from a to z, the digits from 0 to 9, and the hyphen (-)
export _HOSTNAME="pi-home"

# BLOCK DEVICE
#   The SD card or USD SD card reader block device
#   - USB drives will show up as the normal /dev/sdb, /dev/sdc, etc.
#   - MMC/SDcards may show up the same way if the card reader is USB-connected.
#   - Internal card readers normally show up as /dev/mmcblk0, /dev/mmcblk1, ...
#   You can use the lsblk command to get an easy quick view of all block
#   devices on your system at a given moment.
export _BLKDEV="/dev/sdc"

# LUKS ENCRYPTION -------------------------------------------------------------
## Encryption Cypher
export _LUKSCIPHER="aes-cbc-essiv:sha256"

## Encryption Password
export _LUKSPASSWD="password"

## Stage2 regenerate luks uuid
# A value of yes generates a new luks uuid for deployment
export _NEWLUKSUUID="yes"

## Encryption Extra
# On rpi0-1-2-3 you may want to reduce the required memory to unlock
#  _LUKSEXTRA="--pbkdf-memory 131072"
export _LUKSEXTRA=""

# LINUX IMAGE FILE ------------------------------------------------------------
export _IMAGEURL=https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/2022-01-28-raspios-bullseye-armhf-lite.zip
export _IMAGESHA="f6e2a3e907789ac25b61f7acfcbf5708a6d224cf28ae12535a2dc1d76a62efbc"

# PACKAGE ACTIONS -------------------------------------------------------------
export _PKGSPURGE=""
export _PKGSINSTALL="tree htop"

# MINIMAL SSH CONFIG ----------------------------------------------------------
#   Keyfile to be used to access the system remotelly through ssh.
#   Its public key will be added to the system's root .ssh/autorized_keys
export _SSH_LOCAL_KEYFILE="$_USER_HOME/.ssh/id_rsa"

###############################################################################
## Stage 1 Settings ###########################################################

# Custom Stage1 Profile
#   Check functions/stage1profiles.fns for reference. You may instruct hooks
#   here or you may call one predefined stage1profile functions.
#   Optional function:
#   - if stage1_hooks is not defined, you will be prompted
#   - declare it if you want to skip script prompt predefining it
stage1_hooks(){
    stage1profile_complete
}

###############################################################################
## Stage-2 Settings ###########################################################

# Optional stage 2 hooks
#   If declared, this function is called during stage2 build by the
#   stage2-runoptional hook.
#
#   Optional function: can be ommited.
stage2_optional_hooks(){
    myhooks "optional-sys-rootpassword"
}

###############################################################################
##Optional Hook Settings #####################################################

# ROOT PASSWORD CHANGER settings ----------------------------------------------
# Hooks
#   optional-sys-rootpassword
#       Changes the system root password

## The new root password
export _ROOTPASSWD="root_password"

```
P.S.
Mostly of all I am using default config file with changes:
change export _NEWLUKSUUID="yes" to **export _NEWLUKSUUID="no"** in this way have to error like this issue:
[https://github.com/unixabg/cryptmypi/issues/46](https://github.com/unixabg/cryptmypi/issues/46)
If use Your default config file like here:
[https://github.com/unixabg/cryptmypi/blob/master/examples/pios-encrypted-basic-dropbear/cryptmypi.conf](https://github.com/unixabg/cryptmypi/blob/master/examples/pios-encrypted-basic-dropbear/cryptmypi.conf)
That's all ok, but after ``` apt update -y && apt upgrade -y```, I have this error:
[https://github.com/unixabg/cryptmypi/issues/46](https://github.com/unixabg/cryptmypi/issues/46)

**Plz help =)**
And have a nice day @unixabg

unixabg, 2022-02-07:
Greetings,
Thank you for the issue report and detailed information. I am out this week but will get setup to confirm all you submitted soonish. On the upgrading of kernels it can break. In the short term on #46 see comment by gemesa and apply after you upgrade and before reboot on a pristine install.

KevinLage, 2022-02-09:
I had the same problem, try to use this release https://github.com/unixabg/cryptmypi/releases/tag/v4.9-beta

unixabg, 2022-02-23:
Greetings,

Thank you for report and sorry it took me so long to get setup. I do all pristine runs and pristine dev vm for almost every issue. The issue does appear to be reproducible, and I am on the rpi3 builds atm. Please know I will work on resolving this issue. Again thank you for the report.

unixabg, 2022-02-24:
Greetings,
Based on testing dropbear remote unlock with _NEWLUKSUUID="yes" enabled in the conf file does not work, hence this bug. I have changed the default of _NEWLUKSUUID to "no" in next-4.x branch for now.

commit b565216f (2022-02-26): Closes #49. Update unlock.sh for new luks uuid in stage2.
commit 181f7236 (2022-10-28): Closes #49. Update unlock.sh for new luks uuid in stage2.
closed 2022-11-06 by unixabg