unixabg/cryptmypi #50: Unlock and operate without connectivity - interactive session offline

Coqueiro, 2022-08-05:
Hello! First, thanks for the scripts and work; the project is impressive. I have done the manual setup referenced on Kali's blog (https://www.kali.org/blog/secure-kali-raspberry-pi/), and that's how I arrived here.

Secondly: I'm very interested in being able to produce a Kali ARM encrypted environment to run with my RPi4, which is unlockable and operational without having to rely on either wifi or ethernet connection, through an interactive session with a direct keyboard connection, similar to the boot USB drive created through this post here: https://www.kali.org/docs/usb/usb-standalone-encrypted/. That said, I wouldn't mind having the ssh port as an optional way to unlock and operate; it's just not my primary use case.

Is there a particular configuration I can use to achieve this, using one of the examples?

Thank you very much for your attention; I would appreciate any help or guidance, for I'm a little bit lost.

unixabg, 2022-08-09:
Greetings,

So on the usb encryption there is issue #34 , which is sort of where I left that since I focus on the sdcards. However It should be doable to boot from the usb and there might have to be some adjusting in stage#-otherscript.sh scripts. You might try some basic examples and see how far off.

On the sometimes want to unlock with ssh, just do a basic-dropbear example. I think it will just probe over and over until network is alive, so keyboard would be available and upon network ssh should be available.

Report issues you encounter here and we see if we can get you up and running.

Coqueiro, 2022-08-13:
Thanks for the reply. Yeah, I brought up the USB documentation just to exemplify what I was trying to achieve with the SD card + RPi4, I'm not touching a USB at the moment (although I'm really happy with my current encrypted USB).

I have done a bunch of tests with the `kali-complete` example, removing the `optional-sys-sshhub` hook (I noticed sshhub.de is unavailable). Neither `2022.2` nor `2022.1` worked, with slightly different errors. The errors I obtained from trying the newest releases were during the first and second stage, apparently caused by the absence of the re4son config file, even though I believe the re4son version was correctly identified, and I believe I used the proper version for my hardware (I have a Rpi 4B, 8 GB; thus I used `v8l+`). I'm sorry, I forgot to save the error logs, but I believe they are reproducible.

That said, I obtained success using the `kali-linux-2021.3-rpi4-nexmon-arm64.img.xz` release, to the extent that I'm able to unlock the file system and login using an interactive session. However, the SSH connection didn't work, even though I believe I have configured my WLAN configurations accordingly. I even saw logs that Dropbear was able to start, DHCP seems to have attributed a static IP for the RPi, and when running an SSH client, I was getting a password prompt when trying to reach the `root` host.

Although I achieved my starting objective, the project got me interested.  Should the `basic-dropbear` behave differently from the `kali-complete`, regarding WLAN connectivity and being able to unlock the system through an initial SSH connection?

steev, 2022-09-08:
Shouldn't that already be doable?  It sounds like you just want to unlock it with the keyboard?  It should already be possible, as that's the default, adding in ethernet/wifi is just adding an additional way to unlock it.

unixabg, 2022-11-06:
Greetings,
Have you tried the next-4.x branch? It has 2022.3 and it passed basic tests for me.