unixabg/cryptmypi #53: got only blinking cursor (with 2022-09-22-raspios-bullseye-arm64-lite)

thomas725, 2022-10-04:
Hi there!

Amazing that this project exists, loving it!

I've tried to create a profile for the current Pi OS release = "2022-09-22-raspios-bullseye-arm64-lite.img.xz", But all I got when I tried to boot the resulting microSD card was a blinking cursor without any text output on the screen, and nothing on the network.

How can I debug why it failed?

I've created a copy of the"pios-encrypted-basic-dropbear" profile and adapted the cryptmypi.conf file to this (includes generated passwords, but I'll simply generate new ones if I try again & have already reformatted the sd from my first try):
[cryptmypi.conf.txt](https://github.com/unixabg/cryptmypi/files/9703206/cryptmypi.conf.txt)

thomas725, 2022-10-11:
Thanks for the reply!

My first issue was that I tried using cryptmypi with my laptop running Manjaro ≈ Arch Linux as the host system, which failed because it doesn't have a mkinitramfs tool. I think it's replacement here is update-initramfs, but I'm not sure.

After finding that out, I tried instead to build a mostly unchanged copy of "kali-encrypted-basic-nuke" sample with another laptop running Linux Mint 20.3 (Una) ≈ Ubuntu 20.04 (Focal Fossa).
The thing's I've changed:
* entering newly generated luks decrypt & nuke keys, as well as a root pwd
* uncommenting the suggested _PKGSPURGE variable to get a lighter image without desktop
* adding " dstat iftop nethogs" to the _PKGSINSTALL variable
* adapting `_LUKSCIPHER="xchacha20,aes-adiantum-plain64"` & adding `_LUKSEXTRA="--type luks2 --hash sha256 --iter-time 5000 --pbkdf argon2i"` (because https://rr-developer.github.io/LUKS-on-Raspberry-Pi/ shows that this has better performance on the pi4)

My second try got farther then the first. I got a prompt for entering the LUKS passphrase locally, but it doesn't get an IP-adress from my DHCP-server and make an Dropbear ssh server accessible via that, but the physical link works fine, the orange & blue lights where on. The passphrase I generated was too complicated to enter or I had some keyboard layout issue.

I will try the next-4.x branch next.

thomas725, 2022-10-13:
Okey, so my 3rd try was using the pios-encrypted-basic-dropbear-arm64. This time I left all the LUKS settings at their defaults. I only updated the used image to:
```
export _IMAGEURL=https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-26/2022-09-22-raspios-bullseye-arm64-lite.img.xz
export _IMAGESHA="72c773781a0a57160eb3fa8bb2a927642fe60c3af62bc980827057bcecb7b98b"
```
I expected the only 19 days younger version to be very similar, and I already had it laying around from my first try.
Of course I also entered a LUKS keyphrase and a root password, and I also added `dstat iftop nethogs` to the package install list.

With this 3rd try the initramfs environment managed to get an IP-Address from my DHCP server and make the promised Dropbear instance accessible on port 2222. I managed to connect to it, enter the passphrase after getting prompted and getting disconnected again:
```
$ ssh -p 2222 root@IP-Address -i path/to/ssh-key
Enter passphrase for /dev/disk/by-uuid/efa177ce-ad06-4de1-9339-08b6e3c17c9a: 
Connection to IP-Address closed.
```

Sadly though, it didn't manage to boot the system afterwards, failing instead with those errors:
```
Killed
Nothing to read on input.
e2fsck 1.46.2 (28-Feb-2021)
/dev/mapper/crypt: clean, 42073/3993008 files, 649389/155525095 blocks
mount: /sys: sys already mounted or mount point busy.
cat: ´/sys/block/*/mapper/partition´: No such file or directory
fdisk: cannot open /dev/*: No such file or directory
cat: ´/sys/block/*/size´: No such file or directory
Error: Could not stat device /dev/* - No such file or directory.
```

thomas725, 2022-10-13:
So now that I've got a version of 2022-09-22-raspios-bullseye-arm64-lite running on my pi4, I've reran the luks benchmarks on there:

```
root@pios-encrypted-basic-dropbear-arm64:~# cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1       142935 iterations per second for 256-bit key
PBKDF2-sha256     233224 iterations per second for 256-bit key
PBKDF2-sha512     189684 iterations per second for 256-bit key
PBKDF2-ripemd160  120692 iterations per second for 256-bit key
PBKDF2-whirlpool   48046 iterations per second for 256-bit key
argon2i       4 iterations, 236167 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
argon2id      4 iterations, 239062 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b        34.7 MiB/s        33.3 MiB/s
    serpent-cbc        128b               N/A               N/A
    twofish-cbc        128b               N/A               N/A
        aes-cbc        256b        27.1 MiB/s        26.6 MiB/s
    serpent-cbc        256b               N/A               N/A
    twofish-cbc        256b               N/A               N/A
        aes-xts        256b        36.0 MiB/s        34.4 MiB/s
    serpent-xts        256b               N/A               N/A
    twofish-xts        256b               N/A               N/A
        aes-xts        512b        28.9 MiB/s        27.1 MiB/s
    serpent-xts        512b               N/A               N/A
    twofish-xts        512b               N/A               N/A

root@pios-encrypted-basic-dropbear-arm64:~# cryptsetup benchmark -c xchacha20,aes-adiantum-plain64
# Tests are approximate using memory only (no storage IO).
#            Algorithm |       Key |      Encryption |      Decryption
xchacha20,aes-adiantum        256b        59.5 MiB/s        59.7 MiB/s
```

I'm concluding that using `xchacha20,aes-adiantum` would have much better performance than the cypher `aes-cbc-essiv:sha256` that I get when I don't modify the luks variables of the `cryptmypi.conf` file.

unixabg, 2022-11-06:
Greetings,
I have made a couple of commits in the next-4.x branch which fixed issues with later images. I believe the core part of this issue is now resolved and latest PiOS images should pass basic testing. Marking closed for now.

closed 2022-11-06 by unixabg