unixabg/cryptmypi #56: PiOS @ Pi4, after upgrade: only ~5 seconds to unlock

thomas725, 2022-12-04:
Hi there!

I've used your project to setup a Pi4 with an encrpyted PiOS system a few months ago, which worked fine until now. Today I've ran apt upgrade and after [repairing my initramfs](https://unix.stackexchange.com/questions/327914/initramfs-luks-and-dm-mod-cant-boot-after-upgrade) (because the idiot I am and the long timespans in between me doing updates of my servers I've forgot to do it the easy way before rebooting) I now managed to boot my Pi4 again.

But I had to connect a screen to find out why it didn't work initially: Something in the new initramfs timeouts if I don't unlock a few seconds and drops to a shell and prevents the unlock script to boot the system.

So I can still boot if I'm "really fast" at connecting via ssh & entering the keyphrase, but that's a very suboptimal situation, which would prevent me from booting the system from a distance after a power outage or even just rebooting remote if the connection is not perfect enough to allow me to do it fast enough.

Anybody knows how to get rid of this timeout in the initramfs and get back to the previous behaviour (= unlimited time to unlock)?

unixabg, 2022-12-10:
Greetings,
First thank you for the information. I am not sure if this is related to https://github.com/unixabg/cryptmypi/issues/46 which I have left open but have not taken time to examine. Upon next opportunity to examine I will take a closer look at this. When you built image did you use the next-4.x branch? Thanks.

ViRb3, 2023-02-18:
I can suggest my fix from https://github.com/ViRb3/pi-encrypted-boot-ssh:
```sh
sed -i 's/^TIMEOUT=.*/TIMEOUT=100/g' /usr/share/cryptsetup/initramfs/bin/cryptroot-unlock
```

xxxcrow, 2023-08-20:
Any tip on how to change it for a real image? 😁

unixabg, 2023-08-20:
@xxxcrow I would just remaster with the information above.

xxxcrow, 2023-08-20:
@unixabg can you explain what to do please
I bounced of many issues, but this one I can't avoid
I'm trying to do encrypted Kali + initramfs WiFi
And get the same message as the TC, but in my case it's fresh installed, and it's Kali 🫣