unixabg/cryptmypi #57: Fresh build, LUKS password doesn't work; "Cannot initialize device-mapper"

sininspira2, 2022-12-09:
As the title states, first boot after flashing an SD card gets to the "Please unlock disk crypt" prompt, at which point the LUKS password doesn't work and is met with

`Cannot initialize device mapper. Is dm_mod kernel module loaded?`
`Cannot use device crypt, name is invalid or still in use`

Used pios-encrypted-basic example, didn't change anything in the config file except for the device to flash to. Built on Ubuntu 22.04 LTS.

unixabg, 2022-12-09:
Greetings,
Not sure if you did build attempt on master branch or next. I believe what is in next works. If not report back.

sininspira2, 2022-12-19:
Hello,
Sorry for the delay in response. Yes, I made sure I was pulling next-4.x. The LUKS volume is also able to be unlocked if I plug the SD card into another machine.

unixabg, 2022-12-20:
Greetings,
Which raspberry pi are you using? I just spun up a fresh test sd card of pios-encrypted-basic from next-4.x, tested with hardware Pi 3B. All just worked for me.

sininspira2, 2023-01-10:
I tried on a 4 but will try a 3B next

unixabg, 2023-01-10:
Greetings,
What branch are you using? And please know I updated the master branch with a merge from next-4.x on 12/24/2022. I tested _most_ of the images in the next-4.x branch before the merge.

sininspira2, 2023-01-12:
Just pulled the latest from Master and seem to be having the same issue on boot; using pios-encrypted-basic this seems to be the only thing of note in the script output:

> - Calling 7000-stage2-initramfs.hook ...
> Attempting to build new initramfs ... (CHROOT is /mnt/cryptmypi)
>   Creating symbolic links from current physical device to crypttab device (if not using sd card mmcblk0p)
>   Using kernel ''
>   Building new initramfs ...
> W: missing /lib/modules/5.15.0-57-generic
> W: Ensure all necessary drivers are built into the linux image!
> depmod: ERROR: could not open directory /lib/modules/5.15.0-57-generic: No such file or directory
> depmod: FATAL: could not search modules: No such file or directory
> cat: /var/tmp/mkinitramfs_FshBnZ/lib/modules/5.15.0-57-generic/modules.builtin: No such file or directory
> find: ‘/var/tmp/mkinitramfs_FshBnZ/lib/modules/5.15.0-57-generic/kernel’: No such file or directory
> W: Can't find modules.builtin.modinfo (for locating built-in drivers' firmware, supported in Linux >=5.2)
> Adding binary-link /usr/sbin/modprobe
> Adding binary /usr/bin/kmod
>

sininspira2, 2023-01-12:
Going through some other images and it seems ubuntu-encrypted-basic sort of worked? I got "cryptsetup: crypt: set up successfully" and the spinning circle for a while but then it dumped me into initramfs

unixabg, 2023-03-01:
Greetings,
So as of recently, I have had only time to test builds from Kali host vm. So it could be something with ubuntu, not sure. I can get dev environment setup soonish to test again that builds are working from Kali host. Can you share a diff of your kali-encrypted-basic-dropbear/cryptmypi.conf against the one in the master ver 4.11-beta? Please drop any passwords or sensitive info.

alphafox02, 2023-03-01:
I’ll try the diff tonight. I started with next-4 branch before moving to master. Same result in each case, plan to try with kali in a VM tonight. I recall the only change I made to the conf file was to change the luks password and uncomment the line to make the image lighter. I just blew away the conf when doing the git checkout of master and redid the same changes.

unixabg, 2023-03-01:
Greetings,
It is a good test to try building on a Kali VM. Also if using Rpi4 target please read https://github.com/unixabg/cryptmypi/blob/master/examples/kali-encrypted-basic-dropbear/cryptmypi.conf#L16-L23 and adjust accordingly. I have seen where Rpi3 image will not work on Rpi4 and vice versa for some of the examples. Thank you for being willing to help in testing this issue.

alphafox02, 2023-03-01:
Good catch so maybe for the kernel filter I need to just change it to l+ instead of v8+? If so I’ll make the change and rebuild the image one more time with 22.04

alphafox02, 2023-03-02:
Following up, making that one change to the conf file for the pi4 results in everything working perfectly even if the image is built by 22.04. Problem, a least for me, is resolved. Thank you!

unixabg, 2023-03-02:
Greetings,
Thank you for testing and reporting back. I think for now I am going to close this since I can not reproduce the error with a correctly configured conf file to the corresponding hardware platform.

closed 2023-03-02 by unixabg