unixabg/cryptmypi #60: Is this project alive?

xxxcrow, 2023-08-18:
Hi, great project, although I couldn't manage get it working 😁
Not sure if it's still working, and want to know if I should continue trying to set it up and use it the future 🫡

gemesa, 2023-08-18:
Should be working, I have been using it for years. What is your issue?

xxxcrow, 2023-08-18:
@gemesa nice to hear!
I think my last issue was #56, but adding the timeout doesn't do anything (probably I need to rebuild initramfs)(or it's even better to just edit the script)

Also, from minor issues, the WiFi is not always working, and also the build time is pretty long 😅

unixabg, 2023-08-20:
@xxxcrow yes alive.

On changing time rebuild initramfs. Marking completed for this topic.

closed 2023-08-20 by unixabg