unixabg/cryptmypi #62: Unable to unlock Kali dropbear

xxxcrow, 2023-09-09:
Hi, i'm desperately trying to set up an encrypted Kali with a dropbear unlock

After RPI connects to the WiFi and I enter all the passwords, the output looks like that:
(OCR text, not perfect)
```
done.

Begin: Mounting root file system ... Begin: Starting dropbear ... Begin: Running: /scripts/local-top
[ 46.045253]1 cam-dummy-reg: disabling
Killed.
done.

Begin: Running /scripts/local-premount ... done.
Nothing to read on input.
Begin: Waiting for root file system ... Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device. Common problemms:

- Boot args (cat /proc/cmdline)

— Check rootdelay= (did the system wait long enough?)
— Missing modules (cat /proc/modules: ls /dev)
ALERT! PARTUUID=71fc311b-02 does not exist. Dropping to a shell!

BusyBox v1.36.1 (Debian 1:1.36.1-3.1) built-in shell (ash)
Enter "help" for a list of built-in commands.

(initramfs)
```

After editing the script from [edit] and commenting lines 117-120, my output looks the same but with the addition of endlessly repeating lines:
```
Begin: Running /scripts/local-block ... done.
```

unixabg, 2023-09-09:
@xxxcrow , first thank you for the report. Please know that I will attempt to help but it will be some time. Currently I am working on paid projects and will need to finish them before I can be of much help. I think what would be helpful for me (and others who use the project) would be to see your build command and config information with any sensitive information redacted.

xxxcrow, 2023-09-09:
@unixabg nice to hear from you, and nice to know you have some other great things to do 🙏

I thought about adding my config, but i'm not sure what version I should attach

I use examples/encrypted-kali-dropbear with a lot of things uncommented (I will try to add the missing info later, the building machine is just not around at the moment)

There's NEWLUKSUUUD option, which as I understand I need to keep "no"
Many other optional options, confusing

The build command is simple too, nothing extra, something like `` ./cryptmypi examples/encrypted-kali-dropbear ``

And my environment is x86_64 Kali linux