unixabg/cryptmypi #63: Hardcoded __DEBIAN_KERNEL in stage2-otherscript.sh and missing unlock.sh

ph1387, 2024-03-17:
Hi,
first of all: Thank your for the tool. 
I had some issues in the beginning, but I managed to get an encrypted Debian 12 Bookworm installation running on my RaspberryPi 3B+. Took me around four days, but in the end it works. The main reason for this is a hardcoded value for the "__DEBIAN_KERNEL" in https://github.com/unixabg/cryptmypi/blob/master/examples/debian-encrypted-basic-dropbear/stage2-otherscript.sh, which I did not expect. I ran the application multiple times and could not understand, why the changes suggested in https://github.com/unixabg/cryptmypi/issues/56 were not applied in the created initramfs. After changing the "__DEBIAN_KERNEL" to "initrd.img-6.1.0-13-arm64" it worked as expected and replaced the "old" initramfs without the "mv" command failing. I even could revert the suggested changes, as Debian seems to work without them.

So my first question is: Is it possible to change the hardcoded value in 

https://github.com/unixabg/cryptmypi/blob/52227dfaf905ace65fa8555fb3999e9a75b299f5/examples/debian-encrypted-basic-dropbear/stage2-otherscript.sh#L26

to something similar as for example https://github.com/unixabg/cryptmypi/blob/52227dfaf905ace65fa8555fb3999e9a75b299f5/functions/chroot.fns#L134 ? It seems only to be the case for Debian configurations.

My other question is more update related. I ran "apt upgrade" and as many others had to rebuild the initramfs afterwards. So far so good. When unlocking the device the dropbear ssh session failed after I provided the necessary password(s) with a message informing me, that "/etc/unlock.sh" can not be found. I managed to fix this as well by copying parts of the https://github.com/unixabg/cryptmypi/blob/52227dfaf905ace65fa8555fb3999e9a75b299f5/hooks/3000-stage1-setup-encryption.hook#L35 code, that is responsible for creating this file in the first place. What I don't understand is why https://github.com/unixabg/cryptmypi/blob/52227dfaf905ace65fa8555fb3999e9a75b299f5/hooks/7500-stage2-chroot-final.hook#L6 removes it, even though it is needed for actually unlocking the file system via [dropbear](https://github.com/unixabg/cryptmypi/blob/52227dfaf905ace65fa8555fb3999e9a75b299f5/hooks/2700-stage1-dropbear.hook#L45) and thus breaks, at least as far I can tell, any form of remote unlocking if you rebuild the initramfs.

unixabg, 2024-03-18:
@ph1387 - fist thank you for the report. On the stage2-otherscript.sh I think your request is good and I will look in to testing that, on a pi4 only, soonish.

On the removing the I look at that a bit also. As I recall I do not think it was needed once installed, but testing did not, does not currently test upgrading kernels on installs. Upgrading kernels seems to cause some problems so I may need  to examine this a bit as well.

ph1387, 2024-03-18:
@unixabg Great! Sure, I can do that. I still have my notes regarding all the steps I took and will create a separate issue. You are way more experienced with this stuff than I am since I did most of the stuff via trial and error. So there may be some unnecessary steps involved in my notes.

unixabg, 2024-03-18:
@ph1387 - any notes or help on issue is always appreciated. Issues are the best way for me and others to see what issues may happen and how to address them. I intend to update images. If you like put in an issue to update the configs for latest image you want updated, in your case probably the debian images. Thanks again.

commit 5d4e3830 (2024-03-18): Closes #63 Guess kernel version in debian-*/stage2-otherscript.sh.